prusa3d/PrusaSlicer #5483: Travelspeed of wipe-tower / toolchange not adjustable

well-engineered, 2020-12-15:
### Version
2.3.0 Beta3

### Operating system type + version
osX 11.0.1 (20B29) BIG SUR

### 3D printer brand / version + firmware version (if known)
E3D Toolchanger, RRF 2.x

### Behavior
***Travelmoves during a toolchange to the wipetower are very slow.** 
In my case, the tools are picked up outside of the printarea and do perform some preparing actions all done by firmware, when issued a T-command. From there, the printhead needs to move to the wipetower, generated by prusaslicer. This is usually a long move, ~150mm long. The speed of this move is very slow, like extruding speed of ~40mm/s

***To recreate the problem, have the firmware do some preparation-moves outside of the printarea and use the wipetower feature.**

*Expected is a fast move, like the 300mm/s that is set in the option as travel speed OR to have an option to set this speed.

*Actual, when looking at the gCode, there is no F-parameter at all in the mentioned travel move.

Video of the toolchanges in action: https://youtu.be/4G8S2LbDTEQ
Link to the 3mf-file hosted on MS-onedrive: https://1drv.ms/u/s!AjQRNRGFeANNv2TR6z--eHZraP8Q?e=gqp4jq
screenshot of the generated gcode-section in charge for the travel move: [screenshot.zip](https://github.com/prusa3d/PrusaSlicer/files/5695222/screenshot.zip)

*is this a new feature request?
Maybe?

lukasmatena, 2020-12-15:
Just for my reference: this seems to be a one-liner on WipeTower.cpp:1025 (function `toolchange_Change`). It needs some thought and testing to make sure that setting feedrate there will not affect further moves under any scenario.

lukasmatena, 2021-06-03:
This issue should be fixed for 2.4.0 release ( 3459231 ). The move that is described above is now using travel speed as set in Print Settings.

I'll close the issue. Let us know if you consider the issue solved after first public 2.4 alpha is released (in several weeks probably). Thanks.

closed 2021-06-03 by lukasmatena