prusa3d/PrusaSlicer #5485: Hover effects apply even when slicer is backgrounded

t3chguy, 2020-12-15:
### Version
PrusaSlicer-2.3.0-beta3

### Operating system type + version
macOS Catalina Version 10.15.7

### Behavior
Hover effects are applied even when window is not focused/in the foreground.
See video at https://riot.ovh/_matrix/media/r0/download/riot.ovh/gvuawXwyvXSLHsZnCnDgvTcI

###Expectation
Hover effects should only apply to the foregrounded window

FidelCapo, 2020-12-15:
I cannot load your video, but if I understood correctly what you mean, then native applications behave the same way. For example, Finder has active buttons even if it is not focused.

t3chguy, 2020-12-15:
Okay my description is poor, you're right that on Finder hover effects apply even in background. The difference with PrusaSlicer is it even happens when the mouse is within the bounds of the foreground window, unlike Finder.

I'll zip and upload the video, hopefully that helps. See attached.

[WhatsApp Video 2020-12-15 at 14.11.00.mp4.zip](https://github.com/prusa3d/PrusaSlicer/files/5697019/WhatsApp.Video.2020-12-15.at.14.11.00.mp4.zip)

FidelCapo, 2020-12-15:
ok, now i see it. But I can't reproduce it on Big Sur. I tried with more applications.. See the attached video. 
[Screen Recording 2020-12-15 at 17.00.06.zip](https://github.com/prusa3d/PrusaSlicer/files/5697056/Screen.Recording.2020-12-15.at.17.00.06.zip)

t3chguy, 2020-12-15:
Alright, leave it with me, I intend to update to Big Sur this week and will see if that resolves it. Definitely happens regardless of the foreground application currently though. Worked with Quicktime in fg, then multiple apps in bg with PrusaSlicer at the very back.

FidelCapo, 2020-12-15:
We also tested the Catalina version and we also couldn't reproduce the issue. Maybe some unusual monitor and scale settings? Do you have external monitors connected? What kind of apple computer do you use?

t3chguy, 2020-12-15:
`MacBook Pro (16-inch, 2019)`

I am connected to a Samsung CRG9 external monitor.
Dragged PrusaSlicer to external monitor, was unable to reproduce, dragged it back to the laptop display and now could no longer reproduce, how bizarre. I'll try to reproduce it after a reboot.

t3chguy, 2020-12-17:
Can't reproduce it anymore, updating to Big Sur and RC1 shortly. Will re-open if it ever pops back up. Thanks for the rapid debugging!

closed 2020-12-17 by t3chguy

FidelCapo, 2020-12-18:
Ok, thank you.