prusa3d/PrusaSlicer #5487: Painted on supports show up on opposite side of part than where painted

nhendin, 2020-12-15:
### Version
Version 2.3.0 Beta 2, linux 64 GTK3 (AppImage release from github)

### Operating system type + version
ChromeOS Crostini VM container (debian linux)

### 3D printer brand / version + firmware version (if known)
Prusa i3 MK3 stock, FW 3.9.1, Prusament PLA Vanilla White loaded.

### Behavior
Printed my first lithopane (a picture of my 16 year, 8 month old dog at the beach recently). To avoid overhangs on the image side, I tilted the print back a bit as suggested by various internet articles,  I then needed supports on the back side of the print as it would be only held up by a thin line otherwise. 

I used the new painted on supports which are awesome!  However they did something unexpected.
To use a painting analogy, the bleed through to the front of the print. --- that is I painted them on the back, but the surface in teh front (the image for the lithopane) also got supports as well, which I would not have expected with the "paint" metaphor.
It will be more clear from the attached screenshots.

[image]
[image]
[image]

Is there a way to restrict the supports when painted to one side only?

Thank you!

FidelCapo, 2020-12-15:
Does this happen even when using the "Circle"? "The "Sphere" is designed to paint also on sides that are not visible.
[image]

nhendin, 2020-12-15:
Fidel,
That was it!  I didn't see the sphere vs. circle option --- or didn't realize what it was.  Using the circle option works as expected.  

Thank you!

closed 2020-12-15 by nhendin