prusa3d/PrusaSlicer #5488: Add concentric Ironing

Nikoju, 2020-12-15:
### Version
2.3.0-Beta1

### Operating system type + version
Win 10 Home x64

### Behavior
I recently had to print a few rings and would have liked to iron the surface, I've tried to change the top infill pattern but the only possible iron pattern seems to be monolithic... 

In this case, a concentric pattern would have been way nicer, I'd suggest adopting the selected "top fill pattern" for Ironing as well. Or is there a reason only to use line- by line ironing?

Thanking a lot in advance
Nikolaus

_Is this a new feature request?_
YES!

zsolt320i, 2020-12-16:
supporting

stepan-martinek, 2021-01-04:
I desperately need this as well!

sglux, 2022-05-17:
Would really be great.
Nevertheless implementation might be tricky - how to define where the centre is?
It seems so obvious but to find an algorithm that also does something expected on
not-perfectly-circular object or if there is a nose on the circle etc....
Question is: how to tell the slicer where the centre should be. And what, if the object
has multiple circular features...

Nikoju, 2022-05-19:
> Would really be great. Nevertheless implementation might be tricky - how to define where the centre is? It seems so obvious but to find an algorithm that also does something expected on [...]

I'd suggest letting the user pick a center point, ideally by recognizing round contours that can then be selected. I am however not sure whether that's an easy task when talking about stls. nevertheless, I think it should also work fine by just using something similar to the brush tool to select one point on the surface as a center

stepan-martinek, 2022-05-19:
If the technique of user selecting the centre point of a circular ironing is introduced, I would like to see it also for the archimedean chords infill.

stepan-martinek, 2022-05-20:
I suggest defining a centerpoint by selecting 1, 2 or 3 model vertices (2 points are sometimes not enough if the round surface has odd number of edges).
- 1 point selected - use it directly
- 2 points selected - calculate a midpoint between them and use it
- 3 points selected - calculate a midpoint of circumscribed circle of the triangle defined by that 3 points

eximmer, 2022-12-23:
I think it would be best if ironing had an option to match the same pattern as the underlying layer.

Perhaps this could be the default.

_This is overall a really fantatstic slicer.  Absolutely the best slicer in so many ways._

==
PrusaSlicer-2.5.0+linux-x64-GTK3-202209060725.AppImage on Debian Testing

MatchstickWorks, 2023-01-15:
Also supporting this feature request. 

I like the smooth top surfaces you get with ironing, but it would be nice to not have as much post processing needed on circular/round perimeter parts.

datenheim, 2023-01-16:
Yes, would be a great feature, but I don't know how complicated it could get, and what about elliptical perimeters?

MatchstickWorks, 2023-01-16:
Like others have stated, I'm more or less envisioning that Ironing takes the pattern you've specified for your top layers. Want it Monotonic? Iron that way, want it Concentric? Iron concentric. etc etc

You probably wouldn't want Archimedean Chord ironing for square top surfaces because you'd have similar problems to monotonic on circular-like parts.

spin360, 2025-11-27:
This would be so great!

jonathanrswift, 2026-09-02:
This is a reproducible bug, and should not be closed.