prusa3d/PrusaSlicer #5489: Accessibility

bradcla, 2020-12-16:
### Version
2.2.0+

### Operating system type + version
MAC OS 10.15.7

### 3D printer brand / version + firmware version (if known)
Prusa i3 MK3S

### Behavior
Font and colour of font is not positive for people with any vision issues, or anyone in general.  Small lime green font on white in the print settings needs to be customizable.

_Is this a new feature request?_
Yes

YuSanka, 2020-12-16:
Is the same as #4679 and #5162?

YuSanka, 2020-12-16:
It's certainly a bug, which we can't fix currently. But in any case there is a possibility in a PrusaSlicer to customize the colors for UI
[image]

bradcla, 2020-12-16:
That definitely solved the issue for the colours.  Maybe one item that could be changed is the “?”, it looks like a help button versus a settings button.  I have been using the slicer for quite a while and didn’t even see that as anything but help.

Thanks for the quick response,
Brad

[image]<https://user-images.githubusercontent.com/19184891/102321009-643b4580-3f7d-11eb-9c9c-a2ed4953a439.png>

Area5142, 2020-12-17:
I agree. The tooltip points in the direction of a help button:
[image]

Maybe this setting should be moved to preferences where it is expected to be.

bradcla, 2020-12-17:
That was the first place I went when I was looking for the setting.

[Screenshot from 2020-12-17 01-42-42 - Edit]<https://user-images.githubusercontent.com/1320126/102423536-8c708600-4009-11eb-8ad3-435a2034f33d.png>

bubnikv, 2021-08-11:
Settings of colors indicating the difficulty of configuration options (Simple / Advanced / Expert) was moved to Preferences to be easy to discover.

It will be implemented in PrusaSlicer 2.4.0-alpha1.

jfayre, 2022-06-06:
Any progress on this? I just purchased a Prusa 3d mini+ and I am totally blind. As mentioned above, a lot of the UI in Prusa Slicer on Windows isn't accessible. 
Tab navigation is extremely spotty. Also, in the preferences window, none of the checkboxes have proper accessibility labels.

stepikovo, 2022-06-08:
@jfayre this issue was more about bad visibility of some elements (and is already closed), what you are looking for is #7595