prusa3d/PrusaSlicer #5491: Send to Printer temp directory is not created in Chrome OS debian 10 - print fails to start

peterkuykendall, 2020-12-16:
### Version
2.3.0-beta3+linux-x64

### Operating system type + version
Debian 10 - Chrome OS

### 3D printer brand / version + firmware version (if known)
Prusa i3 MK3S - current

### Behavior
Added IP and Key for printer.  It Tests successfully.
When I click the "Send to Printer" button, it never gets to the printer and I get the error 

(prusa-slicer:1435): Gtk-WARNING **: 00:14:08.809: Attempting to store changes into `/home/peter/.local/share/recently-used.xbel', but failed: Failed to create file “/home/peter/.local/share/recently-used.xbel.ONQJV0”: No such file or directory

I've done this on 2 Chrome OS setups and gotten the same error.

I was able to resolve the problem by manually adding the ~/.local/shared directories.  But they should have been added by the app.

_Is this a new feature request?_
no

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

bubnikv, 2020-12-16:
Could you please list the output of
set
command with all your environment variables?

peterkuykendall, 2020-12-16:
Not sure which set command you're asking me to list the results of, but here's the environment variables:

bubnikv, 2020-12-16:
This is not really PrusaSlicer's problem, it is a problem of GTK installation on your Linux on chromebook I suppose.

https://bugzilla.redhat.com/show_bug.cgi?id=693189
https://bugzilla.redhat.com/show_bug.cgi?id=684100

this should be fixed with 
https://gitlab.gnome.org/GNOME/gtk/commit/a50d8ee062dff20262c25eb6e8d9141797a2292e

closed 2020-12-16 by bubnikv