prusa3d/PrusaSlicer #5492: Multi layer filament profiles per print

carcamerarig, 2020-12-16:
Hi
Is it possible to set different different filament profiles (temps, retraction, speeds etc) at multiple layer heights? 
I would like to print PETG on top of TPU and the profiles are quite different. I can set a colour change but see how to change the filament profile mid print. 
Thanks 🙏

rtyr, 2020-12-16:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/3378. You can find possible workaround there.

Closing.

closed 2020-12-16 by rtyr

carcamerarig, 2020-12-16:
> Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/3378. You can find possible workaround there.
> 
> 
> 
> Closing.

Does adding extra extruders and selecting them per layer work, even though there is only one?
Perfect solution I think if it does.

rtyr, 2020-12-16:
Yes. I don't know what printer/firmware are you using, but in general, the firmware should ignore the commands for tool change on single extruder printer.