prusa3d/PrusaSlicer #5503: Physical printer disappears from Slicer upon close of application

pimogo, 2020-12-17:
### Version
Version 2.3.0-beta3+win64

### Operating system type + version
Windows 10 Pro 1909

### 3D printer brand / version + firmware version (if known)
Prusa Original Mk3s 

### Behavior
The behavior i'm seeing consistently on my working copy of the latest beta of prusa slicer is this:  When i add a physical printer on a user defined preset it works while that instance of prusa slicer is open, but when i close the application that physical printer disappears from the slicer. When i go to re-add it and give it the same name i get a message saying that the physical printer exists, yet it doesn't. not in the slicer, anyway. I've tried saving it as snapshot and activating and importing a config bundle w/ physical printer (after saving it, of course) but to no avail.  the only solution is to re-add my octoprint credentials each time i reopen prusa slicer? 

is this supposed to be like this?

FidelCapo, 2020-12-18:
Unfortunately, we are unable to reproduce this error. Can you please attach your configuration folder (Help->Show Configuration Folder)? Can you make a video with precise steps to reproduce this issue?
Thank you.

pimogo, 2020-12-18:
i believe i figured it out myself. my user preset had a space in front of it (to sort to the top). when i try creating a physical printer off of another preset it works fine. i tested removing the space from my user preset and then it worked. 

sorry for the false alarm.. also i notice w/ the latest beta leading spaces are no longer allowed when saving presets.

lukasmatena, 2020-12-19:
Thanks for the investigation. Yes, it should no longer be possible to use names with leading/trailing spaces since beta2 to avoid issues like this. That probably makes this issue solved, so I'll close it.

fila612, 2021-02-24:
It's still possible to reproduce this error in the final version:
I've named my user presets as "_preset_name = MMU2S - Prusa i3 MK3S (Flow100; Thumbnails);MMU2S Single - Prusa i3 MK3S (Flow100;_ " and the behvaiour was exactly like write above.
After changing the names in user presets to shorter versions like "mmu2s" and "mk3s" its working.
maybe there is still a litte bug inside ;)

lukasmatena, 2021-02-24:
I can confirm that the issue appears when the preset is named "MMU2S - Prusa i3 MK3S (Flow100; Thumbnails)". Naming the preset "MMU2S Single - Prusa i3 MK3S (Flow100; " is not possible due to the trailing space, which PS2.3.0 refuses to do.

It appears that it is the semicolon that poisons the preset. And that loading from the INI is fine and bad things start happening in `PlaterPresetComboBox::update()`. Maybe @YuSanka would see more.

YuSanka, 2021-02-25:
Thanks for your report. The problem was related to use of ";" in preset name. But it wasn't a problem in the feature.
This bug is fixed and will be part of next release.

closed 2021-02-25 by YuSanka