prusa3d/PrusaSlicer #5504: Prusa Slicer makes so many unnecessary movements compare to Cura

Wextor, 2020-12-17:
**### Version PrusaSlicer**
Version:   2.3.0-beta3+win64
Build:     PrusaSlicer-2.3.0-beta3+win64-202012121809

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 19041), 64-bit edition
Total RAM size [MB]: 17,133MB
OpenGL installation
GL version:   4.6.14757 Compatibility Profile Context 20.11.2 27.20.14501.18003
Vendor:       ATI Technologies Inc.
Renderer:     Radeon (TM) RX 480 Graphics
GLSL version: 4.60

**### 3D printer brand / version + firmware version (if known)**
Creality Ender 3 4.7.2 Marlin

**### Behavior**
* _Describe the problem_
Prusa Slicer makes so many unnecessary movements compare to Cura

* _Steps needed to reproduce the problem_
Add any model with walls like that one in picture

* _Expected Results_
Movements should be inside infill to prevent blobs from not fully retracted filament

* _Actual Results_
Difference between Cura and Prusa Slicer
[image]

And how its look after print
[image]

_Is this a new feature request?_
No idea, maybe this is a bug, but if not - yes this is a new feature request

#### Project File (.3MF) where problem occurs
[Project.zip](https://github.com/prusa3d/PrusaSlicer/files/5712240/Project.zip)

hejllukas, 2020-12-17:
You probably uploaded a different 3MF than what is shown in the first picture. But you could try enabling `Avoid Crossing Perimeters`. We reworked this feature in 2.3.0-beta3, and it should force most of the travels inside one object to lead inside an infill or perimeters (but not crossing perimeters), which may help with your issue.

Wextor, 2020-12-17:
@hejllukas Sorry, you're right, I edited the post and now project is ok, I tried this option many times and I checked different seams but this not resolving the problem. Could you look at this project again?

hejllukas, 2020-12-17:
Thank you, I looked at your project, and you have set `Avoid crossing perimeters - Max detour length` to `1`. This means that `Avoid crossing perimeters` is used only if a detour (new travel wichs follows a perimeter or is inside an infill to minimize crossing of external perimeter) is shorter or equal to 1mm. If you set it to zero, then it disables this limit. Then most of the travels are inside the object, as shown in the following picture. But these travels could be quite long.
[image]

Wextor, 2020-12-17:
Resolved

closed 2020-12-17 by Wextor