prusa3d/PrusaSlicer #5505: [Linux][GTK3] Dropdowns in quick settings are cut vertically with some themes

jeanblanchard, 2020-12-18:
### Version

Version 2.3.0-rc1+linux-x64-GTK3 (but this is not new to this version)

### Operating system type + version

* Linux (Solus 4.1)
* KDE Plasma 5.20.3
* GTK 3.24.24
* GTK Theme: Breeze

### Behavior

The text in the print settings, filament and printer dropdowns, in the main settings panel (the ones on the "Panel" tab) is cut vertically, with some GTK3 themes (notably all Breeze-based themes)

The "Supports" and "Infill" dropdowns, which seem to use a different widget, are not impacted.

The Preset dropdown in the other settings tabs is fine too. It appears to use the same widgets as the first 3, but has a greater height.

My guess is, there is probably a maximum height applied on these 3 widgets, which would explain why all themes are not impacted.

### Screenshots

This is the issue:
[image]

This is fine:
[image]

nordurljosahvida, 2021-01-03:
After the fixes of https://github.com/prusa3d/PrusaSlicer/issues/4135 https://github.com/prusa3d/PrusaSlicer/issues/4135 I just noted instead some horizontal cuts in 2.3.0-rc2 GTK3 appimage under Ubuntu 20 HiDPI screen.

[image]

[image]

This happens with multiple printers and filaments but does not affect all, cannot find a pattern for now. Print profile box is unaffected.

bubnikv, 2021-01-04:
Do you have any filament and printer profile with an excessively long name?
Maybe the GTK widget formats according to the longest text in the combo box?

nordurljosahvida, 2021-01-05:
Aha, that's right. I have some physical printer printer names that are longer, same things with filaments. Should we enforce a left-justification of text when the menu isn't open?

bubnikv, 2021-01-05:
BTW how does the combo box look like when rolled down? Are the texts clipped as well?

bubnikv, 2021-01-05:
I asked at wx-users list. I am not sure whether it is a wxWidgets issue or
a GTK issue.

nordurljosahvida, 2021-01-05:
Thanks for looking into this.

When open the menu overflows into the left and right [therefore off-screen, or at least until the very edge] so the text isn't clipped.

[image]
[image]

vintagepc, 2021-07-04:
I see some commits so I'm guessing it may have already been fixed but I'll add in that I'm also observing this on KDE with a similar dark theme:
[image]

jeanblanchard, 2021-07-08:
I can confirm the issue is fixed in PrusaSlicer 2.3.2, released today :tada: 

Thanks @vintagepc for bringing it up right in time in the RC phase for the fix to be backported before the release :+1: 

[image]

closed 2021-07-08 by jeanblanchard

vintagepc, 2021-07-08:
🎉 whoo, update! 😀