prusa3d/PrusaSlicer #5506: Problem with connection to Octoprint using SSL (HTTPS)

rufik, 2020-12-18:
### Version
PrusaSlicer 2.2.0+win64
Octoprint 1.5.2 `{"api":"0.1","server":"1.5.2","text":"OctoPrint 1.5.2"}`

### Operating system type + version
Windows 10 Pro 20H2 compilation 19042.685
```
PrusaSlicer
Version:   2.2.0+win64
Build:     PrusaSlicer-2.2.0+win64-202003211145
Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 19042), 64-bit edition
Total RAM size [MB]: 16,964MB
OpenGL installation
GL version:   4.6.0 - Build 27.20.100.8280
Vendor:       Intel
Renderer:     Intel(R) UHD Graphics 630
GLSL version: 4.60 - Build 27.20.100.8280
```

### 3D printer brand / version + firmware version (if known)
Ender-3 Pro, firmware TH3D 2.12b

### Behavior
**Connection to Octoprint using HTTP (SSL) does not work**. I have Linux box (Ubuntu Focal armhf) running successfully Octoprint 1.5.2. I'd like to migrate from Cura to PrusaSlicer, so I've made simple printer configuration and I put following setting for "Print Host Upload" section:
```
Host Type: Octoprint
Hostname: https://octo
API Key: [...]
```
So it's really simple. I'm running Octoprint at https://octo dns name, it's resolvable and it uses my self-signed SSL certificate. CA for cert is already loaded into Windows certs storage, so Chrome (and other web browsers) has no problem with that at all:
[image]
Cura does work as well. 

So when I press "Test" button I'm getting no-meaning message:
```
Could not connect to OctoPrint: 
Note: OctoPrint version at least 1.1.0 is required.
```
[image]

I'm not able to locate any log file for PrusaSlicer, so I'm unable to verify this issue deeper.
How to debug what's going on?

rufik, 2020-12-22:
OK, I've found that I can run `prusa-slicer-console.exe --loglevel 9` and gather some meaningful logs :)
Here they are:
```
[2020-12-22 11:27:49.454782] [0x00000cc8] [info]    OctoPrint: Get version at: https://octo/api/version
*   Trying 192.168.aa.bb...
* TCP_NODELAY set
* Connected to octo (192.168.aa.bb) port 443 (#0)
* schannel: SSL/TLS connection with octo port 443 (step 1/3)
* schannel: checking server certificate revocation
* schannel: ALPN, offering http/1.1
* schannel: sending initial handshake data: sending 190 bytes...
* schannel: sent initial handshake data: sent 190 bytes
* schannel: SSL/TLS connection with octo port 443 (step 2/3)
* schannel: failed to receive handshake, need more data
* schannel: SSL/TLS connection with octo port 443 (step 2/3)
* schannel: encrypted data got 3547
* schannel: encrypted data buffer: offset 3547 length 4096
* schannel: next InitializeSecurityContext failed: Unknown error (0x80092012) - The revocation function was unable to check revocation for the certificate.
* Closing connection 0
* schannel: shutting down SSL/TLS connection with octo port 443
* schannel: clear security context handle
[2020-12-22 11:27:49.716445] [0x00000cc8] [error]   OctoPrint: Error getting version: SSL connect error:
schannel: next InitializeSecurityContext failed: Unknown error (0x80092012) - The revocation function was unable to check revocation for the certificate.
[Error 35], HTTP 0, body: ``
```

So it looks like a problem with usage of WinSSL (schannel) function to check revocation list. **Can this function be disabled somehow?**

bubnikv, 2021-01-05:
I am not sure. We are using https://curl.se/libcurl/c/ for all TCPIP traffic. On Windows libcurl is configured to use the Microsoft SSL stack.

nerxses, 2021-02-22:
can confirm this as well. 

Octoprint is running with an custom certificate and i am getting the same errors with Win64 Pro. (System is on german but it's the same)
```
[2021-02-22 22:11:34.611362] [0x00002720] [info]    OctoPrint: Get version at: http://10.42.0.33/api/version
*   Trying 10.42.0.33...
* TCP_NODELAY set
* Connected to 10.42.0.33 (10.42.0.33) port 80 (#0)
> GET /api/version HTTP/1.1
Host: 10.42.0.33
User-Agent: PrusaSlicer/2.3.0+win64
Accept: */*
X-Api-Key: AB085A571333455BB669A8E5578D1512

< HTTP/1.1 302 Found
< Cache-Control: no-cache
< Content-length: 0
< Location: https://10.42.0.33/api/version
<
* Connection #0 to host 10.42.0.33 left intact
* Issue another request to this URL: 'https://10.42.0.33/api/version'
*   Trying 10.42.0.33...
* TCP_NODELAY set
* Connected to 10.42.0.33 (10.42.0.33) port 443 (#1)
* schannel: SSL/TLS connection with 10.42.0.33 port 443 (step 1/3)
* schannel: checking server certificate revocation
* schannel: using IP address, SNI is not supported by OS.
* schannel: ALPN, offering http/1.1
* schannel: sending initial handshake data: sending 171 bytes...
* schannel: sent initial handshake data: sent 171 bytes
* schannel: SSL/TLS connection with 10.42.0.33 port 443 (step 2/3)
* schannel: failed to receive handshake, need more data
* schannel: SSL/TLS connection with 10.42.0.33 port 443 (step 2/3)
* schannel: encrypted data got 1443
* schannel: encrypted data buffer: offset 1443 length 4096
* schannel: next InitializeSecurityContext failed: Unknown error (0x80092012) - Die Sperrfunktion konnte keine Sperrprüfung für das Zertifikat durchführen.
* Closing connection 1
* schannel: shutting down SSL/TLS connection with 10.42.0.33 port 443
* schannel: clear security context handle
[2021-02-22 22:11:34.719459] [0x00002720] [error]   OctoPrint: Error getting version: SSL connect error:
schannel: next InitializeSecurityContext failed: Unknown error (0x80092012) - Die Sperrfunktion konnte keine Sperrprüfung für das Zertifikat durchführen.
[Error 35], HTTP 0, body: ``
```

dani190, 2021-03-13:
Seeing this issue also. I have a personal CA and the certs all show as trusted in my browser.

[image]

rufik, 2021-03-15:
> I am not sure. We are using https://curl.se/libcurl/c/ for all TCPIP traffic. On Windows libcurl is configured to use the Microsoft SSL stack.

@bubnikv So it looks like CURLSSLBACKEND_SCHANNEL option is used (https://curl.se/libcurl/c/curl_global_sslset.html).
I think it's possible to affect revocation function via these libcurl options: `CURLSSLOPT_NO_REVOKE` or `CURLSSLOPT_REVOKE_BEST_EFFORT`: https://curl.se/libcurl/c/CURLOPT_SSL_OPTIONS.html
I'd try myself but I'm out of free time unfortunetly...but if someone can compile with this libcurl options enabled I'd be happy to try it out :)

bubnikv, 2021-04-07:
> So it looks like CURLSSLBACKEND_SCHANNEL option is used (https://curl.se/libcurl/c/curl_global_sslset.html).
I think it's possible to affect revocation function via these libcurl options: CURLSSLOPT_NO_REVOKE or CURLSSLOPT_REVOKE_BEST_EFFORT: https://curl.se/libcurl/c/CURLOPT_SSL_OPTIONS.html
I'd try myself but I'm out of free time unfortunetly...but if someone can compile with this libcurl options enabled I'd be happy to try it out :)

That may work on Windows. We are noobs in SSL certificates. What are the security risks connected with such configuration? Should we make it an option to the user?

bubnikv, 2021-04-07:
BTW we are having similar issues on OSX.

rufik, 2021-04-07:
> That may work on Windows. We are noobs in SSL certificates. What are the security risks connected with such configuration? Should we make it an option to the user?

It's about CRL (https://en.wikipedia.org/wiki/Certificate_revocation_list) - list of certs that has been revoked and are no longer valid. So it's an issue that kind you still trust revoked ssl certs.
I think the best is to give user choice - a combobox with options: _Use CRL_ (default), _Try to use CRL_ and _Do not use CRL_ accordingly to libcurl options: _default_, _CURLSSLOPT_REVOKE_BEST_EFFORT_ and _CURLSSLOPT_NO_REVOKE_.

bubnikv, 2021-04-07:
The SSL channel on Windows we use is the Windows provided SChannel. Now why is SChannel unable to check revocation for the certificate? Because it is self signed?

Could we disable checking for certificate revocation just for self signed certificate?

> * schannel: next InitializeSecurityContext failed: Unknown error (0x80092012) - The revocation function was unable to check revocation for the certificate.

jay, 2021-04-08:
0x80092012 is [CRYPT_E_NO_REVOCATION_CHECK](https://docs.microsoft.com/en-us/windows/win32/api/wincrypt/nf-wincrypt-certverifyrevocation). That can be ignored by [SCH_CRED_IGNORE_NO_REVOCATION_CHECK](https://docs.microsoft.com/en-us/windows/win32/api/schannel/ns-schannel-schannel_cred) which is set by [CURLSSLOPT_REVOKE_BEST_EFFORT](https://curl.se/libcurl/c/CURLOPT_SSL_OPTIONS.html). You can read the details of how it came about in curl/curl#4981.

Ignoring that error is not the default since the error is not specific to certificates without revocation points, which means it's tantamount to soft revocation check and curl's default is hard revocation check.

bubnikv, 2021-04-08:
@jay Thanks, we can try that.

BTW, why don't you guys generate signed certificates with https://letsencrypt.org/

rufik, 2021-04-08:
Letsencrypt requires web page to be exposed and reachable in public. It's not my case - I have Octoprint only in my LAN (and some others services as well). So I'm using self-signed certs (with IP & DNS Name in SAN) and I trust to my CA - it just works fine.

kocikdav, 2021-04-27:
Hello @rufik @dani190 @nerxses and anybody having issues with self-signed certificate for Octoprint on Windows. Can I ask you to test this build? There should be checkbox in Physical Printer dialog saying "Ignore certificate revocation checks", please try upload with this option checked. Please note that this build is experimental and other features might broken / different than release or latest master.
https://www.dropbox.com/s/ilzgpq1593k0737/PrusaSlicer-dk_curl.rar?dl=0

rufik, 2021-04-28:
I've just checked - using https and Test button - and it seems to be OK when _"Ignore certificate revocation checks"_ option is checked :) Very nice, thanks you. That's the solution we need.

[image]

kocikdav, 2021-04-29:
@rufik Great! Would you please verify that actual upload of gcode is working as well?

bubnikv, 2021-04-29:
> I've just checked - using https and Test button - and it seems to be OK
when *"Ignore certificate revocation checks"* option is checked :) Very
nice, thanks you. That's the solution we need.

Now the bonus question: Why is this needed? What's wrong with the
certificate? What's wrong with Microsoft Windows TLS implementation?

rufik, 2021-04-29:
@kocikdav Uploading to octoprint works fine as well :)

dani190, 2021-05-26:
Just wanted to see if this will be merged in soon? I would download the build you noted but you said it could be unstable in other ways.

marcobox, 2021-06-18:
Is this checkbox working with OSX too? I am having similar issues and would like to try the experimental build on the Mac to see if it works.

ssill2, 2021-06-18:
I would imagine with any use of SSL/TLS you're going to have to deal with the issue of checking of the CN in the certificate as well as whether the slicer trusts the issuer of the cert being presented by the octo-print server.

bubnikv, 2021-06-20:
: Is this checkbox working with OSX too? 

Sorry, there is no such option offered by OSX API. The OSX API is purely documented in how it handles these self-signed certificates. Likely they would only work if we bundled your self-signed certificate as part of the application and the application was signed, but it is only my wild guess based on some hints skimmed from the OSX API documentation. If we wanted to "fix" this issue, we would likely have to revert to the OpenSSL implementation on OSX, circumventing the native OSX API.

ssill2, 2021-06-20:
Yes I totally understand.  I was more making folks aware that using TLS requires you to do a number of back-flips in order to use self-signed certs.   Usually the OS can be made to trust the cert by installing it in the system's truststore.

bubnikv, 2021-06-20:
@ssill2 You would not believe how much time @kocikdav and me have wasted on this topic. We even asked the author of libcurl. The answer was something along the lines that the native TLS implementations are weird beasts.

> Usually the OS can be made to trust the cert by installing it in the system's truststore.

That certainly did not help.

It may be that some of our very old slicer builds compiled libcurl using OpenSSL TLS implementation, not the OSX native TLS implementation and that worked for self signed certificates. We may try that one day if there is enough of interest.

ssill2, 2021-06-20:
yep, I've had to do a LOT of stuff with TLS for my job.  Thankfully with java it's fairly easy to make self-signed certs work by simply adding the key to cacerts, java's default truststore, and it's possible to override hostname verification if the need should arise.   I've never done this in C/C++.  I don't currently use octoprint but I appreciate the pain that TLS is :)

ssill2, 2021-06-20:
not to mention getting into the ciphers etc and supporting things like TLS 1.3 etc lol

ssill2, 2021-06-20:
Here's a run-down of a normal TLS conversation to illustrate why self-signed certs don't work by design out of the box.
https://www.cloudflare.com/learning/ssl/what-happens-in-a-tls-handshake/

Most os/browser vendors distribute a bundle of CA certs that's updated fairly frequently.  This bundle is what the os uses to evaluate trust of tls certificates.  This would include the big guys like GoDaddy, Network Solutions etc.  While you CAN often override the truststore, or add to it, to use self-signed certs, this would be OS specific.  I know you can with curl pass I think a -k in order to accept certs from untrusted certificate authorities.  Besides this trust check the two ends of the conversation have to agree on what version of TLS to use(1.0, 1.1, etc) as well as encryption ciphers.  The last check that's on by default in TLS is a hostname check.   This checks that the canonical name(CN) in the cert matches what hostname you're connecting to.   So if you're connecting by ip, but the server prints a cert with a CN that is a hostname, you'll likely have a failed connection.  The easiest way to get around this is to have an /etc/hosts entry that matches the name in the CN of the certificate.

I think most OSes allow you to import the self-signed cert into the trust store, similar to the prompt a browser might give you to allow you to trust the cert.   This would definitely have to be researched per platform.

sebastian-be, 2021-06-28:
I've had same issue as reported above with v2.3.1. I was able to resolve it by regenerating the server certificate, for OctoPrint, with the ssl extension "crlDistributionPoints" set.

Using easyrsa, change the x509-types/COMMON:
uncomment
#crlDistributionPoints = URI:http://<server>/<path>/ca.crl

and piont the URL to where the certificate revokation list is hosted. CRL can be generated using:

./easyrsa gen-crl
# + convert to DER format
openssl crl -in /.../CA/pki/crl.pem -outform DER > /.../www/ca/ca.crl

ostat, 2021-09-04:
I can confirm that this issue is resolved in **PrusaSlicer 2.4.0-alpha1**. 
The "Ignore certificate revocation checks" has been added.

bubnikv, 2021-09-06:
@ostat Thanks for heads up. Happily closing.

closed 2021-09-06 by bubnikv

allanonmage, 2024-01-17:
Seems the ignore button doesn't work in 2.7.1.  I get the error with or out without changing the ignore setting.  I was able to work around it be switching to http.  It's on my LAN, so this is an OK workaround for now.  

Verbose error message:
```
Could not connect to OctoPrint: SSL peer certificate or SSH remote key was not OK:
schannel: SEC_E_UNTRUSTED_ROOT (0x80090325) - The certificate chain was issued by an authority that is not trusted.
[Error 60]
Note: OctoPrint version at least 1.1.0 is required.
```

From Help, About:

```
PrusaSlicer
Version:   2.7.1+win64
Build:     PrusaSlicer-2.7.1+win64-202312121425

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 19045), 64-bit edition
Total RAM size [MB]: 68,607MB
OpenGL installation
GL version:   4.6.0
Profile:      Core
Vendor:       NVIDIA Corporation
Renderer:     NVIDIA GeForce RTX 3080/PCIe/SSE2
GLSL version: 4.60.0
Textures compression:       Enabled
<details>
<summary>Installed extensions:</summary>
GL_AMD_multi_draw_indirect
GL_AMD_seamless_cubemap_per_texture
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_compute_variable_group_size
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
[... 340 lines cut ...]
GL_NV_uniform_buffer_unified_memory
GL_NV_vertex_array_range
GL_NV_vertex_array_range2
GL_NV_vertex_attrib_integer_64bit
GL_NV_vertex_buffer_unified_memory
GL_NV_vertex_program
GL_NV_vertex_program1_1
GL_NV_vertex_program2
GL_NV_vertex_program2_option
GL_NV_vertex_program3
GL_NV_viewport_array2
GL_NV_viewport_swizzle
GL_OVR_multiview
GL_OVR_multiview2
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_lod
GL_SGIX_depth_texture
GL_SGIX_shadow
GL_SUN_slice_accum
GL_WIN_swap_hint
WGL_EXT_swap_control
</details>

```

jamiekin, 2024-07-28:
This issue is back in 2.8.0.  The temporary workaround of removing the redirect line of haproxy.cfg and switching from https to http allows me to print from PrusaSlicer. Is there a real fix for this problem?