prusa3d/PrusaSlicer #5507: Crash on Eject in Second Instance on macOS

b-fitzpatrick, 2020-12-18:
### Version
2.3.0-rc1

### Operating system type + version
macOS Big Sur 11.1

### 3D printer brand / version + firmware version (if known)
Prusa MINI

### Behavior
* _Describe the problem_
When ejecting a USB drive after saving gcode in a second instance of PrusaSlicer, it crashes with "Internal error: wait error: No child processes".

* _Steps needed to reproduce the problem_
Open PrusaSlicer. Open a second instance using the "Open new instance" command from the Dock. In the second instance, open a model, slice the model, save the gcode to USB drive, and click eject in the save notification.

If I open a second instance with the command "open -a /Applications/PrusaSlicer.app --new", this crash does not occur.

* _Expected Results_
No crash.

* _Actual Results_
Crash. The USB drive is ejected.

* _Screenshots from __*PrusaSlicer*__ preview are preferred_
[image]

* _Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
[calicat.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/5716108/calicat.3mf.zip)

lukasmatena, 2020-12-22:
Thanks for the report, should be fixed in [2.3.0-rc2](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.3.0-rc2) (99f5dfb). It is possible that the application will report the eject as failed, even though it would actually have succeeded. It should no longer crash though.

Can you please confirm that?

b-fitzpatrick, 2020-12-23:
Thanks. I confirmed that with the same steps, the drive is ejected but reported as failed.

lukasmatena, 2020-12-23:
Ok. Hopefully we'll improve it for 2.4 release.

bubnikv, 2021-01-07:
The issue is described in here: https://jmmv.dev/2008/10/boostprocess-and-sigchld.html

commit 2b02747e (2021-01-07): On OSX, we use boost::process::spawn() to launch new instances of PrusaSlicer from another PrusaSlicer.
boost::process::spawn() sets SIGCHLD to SIGIGN for the child process, thus if a child PrusaSlicer spawns another
subprocess and the subrocess dies, the child PrusaSlicer will not receive information on end of subprocess
(posix waitpid() call will always fail).
https://jmmv.dev/2008/10/boostprocess-and-sigchld.html
The child instance of PrusaSlicer has to reset SIGCHLD to its default, so that posix waitpid() and similar continue to work.
Fixes #5507
closed 2021-01-07 by bubnikv