prusa3d/PrusaSlicer #5510: ctrlsub.cpp(231): assert "IsValid(n)" failed in GetClientData(): Invalid index passed to GetClientData()

pmjdebruijn, 2020-12-18:
### Version
` 2.3.0~rc1+12~g2fa582802`

### Operating system type + version
Xubuntu 20.04

### Behavior
When switching between two printers in the right panel, I get the backtrace added below. I'm guessing this is a fairly recentish regression, as I recall switching between two printers without any issues not so long ago.

Note, I do not have any issues/backtraces when switching between the Filaments or Presets.

```
ASSERT INFO:
../src/common/ctrlsub.cpp(231): assert "IsValid(n)" failed in GetClientData(): Invalid index passed to GetClientData()

BACKTRACE:
[1] wxItemContainer::GetClientData(unsigned int) const
[2] wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&)
[3] wxEvtHandler::SearchDynamicEventTable(wxEvent&)
[4] wxEvtHandler::TryHereOnly(wxEvent&)
[5] wxEvtHandler::ProcessEventLocally(wxEvent&)
[6] wxEvtHandler::ProcessEvent(wxEvent&)
[7] wxEvtHandler::SafelyProcessEvent(wxEvent&)
[8] wxControlWithItemsBase::SendSelectionChangedEvent(int)
[9] g_signal_emit_valist
[10] g_signal_emit
[11] gtk_combo_box_set_active_iter
[12] g_cclosure_marshal_VOID__STRINGv
[13] g_signal_emit_valist
[14] g_signal_emit
[15] g_closure_invoke
[16] g_signal_emit_valist
[17] g_signal_emit
[18] gtk_widget_activate
[19] gtk_menu_shell_activate_item
[20] g_signal_emit_valist
[21] g_signal_emit
[22] gtk_main_do_event
[23] g_main_context_dispatch
[24] g_main_loop_run
[25] gtk_main
[26] wxGUIEventLoop::DoRun()
[27] wxEventLoopBase::Run()
[28] wxAppConsoleBase::MainLoop()
[29] wxEntry(int&, wchar_t**)
[30] __libc_start_main
```

lukasmatena, 2020-12-19:
Thanks. Am I right in assumption that it does not happen with builds from our buildserver, but only when linked to wx3.0?

pmjdebruijn, 2020-12-19:
When I build against 3.1.3-patched (built locally outside of the Prusa deps build scripts), I still have this issue.

It seems the Prusa deps build script explicitly turn off assertions:
https://github.com/prusa3d/PrusaSlicer/blob/master/deps/wxWidgets/wxWidgets.cmake#L29
https://docs.wxwidgets.org/3.0/group__group__funcmacro__debug.html#ga7ce424d27d39bb92c26c80b6a1903725

But presumably those assertions are there for a reason, and the wx folks have consciously enabled them by default.

So this issue isn't related to wx3.0 vs 3.1.3-patched, your internal library builds are preventing these regression from being noticed during development.

While one might argue whether assertions should be enabled for production builds (even though the wx folks think so), during develop you should consider enabling assertions.

As I already noted earlier, this issue was likely introduced semi-recently.

pmjdebruijn, 2020-12-19:
This may offer a short term workaround:
https://docs.wxwidgets.org/3.0/group__group__funcmacro__debug.html#ga36a77b1afdfcac5d30beeb7c727b797c

pmjdebruijn, 2020-12-19:
I just tested that adding `wxDisableAsserts();` to `GUI_App::OnInit()`, seems to at least suppress these assert messages, until a proper fix can be worked out.

YuSanka, 2021-01-05:
I can't reproduce your assert. I tried to debugging on Ubuntu 20.04, set brake point to the ../src/common/ctrlsub.cpp(231) and controlled the value checking, but index is valid every time for me.

pmjdebruijn, 2021-01-05:
#5152 is probably indirectly related

pmjdebruijn, 2021-01-05:
> I can't reproduce your assert. I tried to debugging on Ubuntu 20.04, set brake point to the ../src/common/ctrlsub.cpp(231) and controlled the value checking, but index is valid every time for me.

You did build PrusaSlicer against an assert enabled build of wxGTK right? The Prusa wxGTK builds have asserts entirely disabled at build time, thus cannot produce any assert messages.

What can I do to check things on my end? Are there any particular variables of relevance that I can printf to better see what happening on my end?

kylian-002, 2021-01-18:
got the same problem. i build it from the AUR on arch linux this has always worked probaly at some point a update has broke it as i wanted to use it just now and am hit with the same problem

YuSanka, 2021-02-18:
> You did build PrusaSlicer against an assert enabled build of wxGTK right? The Prusa wxGTK builds have asserts entirely disabled at build time, thus cannot produce any assert messages.

Yes, I enabled wxWidgets asserts, but still don't see this assert.

> What can I do to check things on my end? Are there any particular variables of relevance that I can printf to better see what happening on my end?

Can you be more specific? Which printers do you use in the printers presets list .

YuSanka, 2021-02-24:
@pmjdebruijn, Do you still observe this issue?

pmjdebruijn, 2021-02-24:
Yes, but I've observed not having the issue on another machine. After playing around a bit more, the issue seems to go away after selecting one more machine from the list. So this issue may very well only occur when having three machines selected (and no custom ones).

I hope you'll be able to reproduce using the attached PrusaSlicer.ini:
[GitHubIssue5510_PrusaSlicer.zip](https://github.com/prusa3d/PrusaSlicer/files/6037879/GitHubIssue5510_PrusaSlicer.zip)

YuSanka, 2021-02-25:
@pmjdebruijn Thanks for your investigation. Your INI really helped me figure out what was going on. 
Please retest this fix and fix for the #5152. And if they are fixed, close them.

bubnikv, 2021-08-06:
Fixed in PrusaSlicer 2.4.0-alpha1.

closed 2021-08-06 by bubnikv