prusa3d/PrusaSlicer #5511: Sequential tool path ordering of multiple parts within the same object, aka. perimeter tracking

bartrv, 2020-12-18:
### Version
2.3.0-beta2 Win64

### Operating system type + version
Windows 10 x64

### 3D printer brand / version + firmware version (if known)
Prusa MK3S

### Behavior
Currently the travel ordering for MMU isolated path clusters/groups around the perimeter of an object are calculated based on an x/y scan instead of a proximity or perimeter tracking adding unnecessary additional time through multiple extended travel moves. See below:
            1   2   3
4                               5
6                               7
            8   9  10

Specifically, when MMU printing windows in an architectural model, the window material is deposited for the 1st window starting at on end, then jumps all the way across the model for the next window and then back to the other side. Every retract/move sequence can take about a full second to traverse a larger model.  For models with hundreds disconnected perimeter objects and layers, this adds a lot of time to the print.

_Is this a new feature request?_
Yes, could a perimeter tracking optimization option be added to order the tool path start locations for multiple parts in the same object sequentially around the perimeter. See below:
             1   2   3
10                              4
 9                               5
             8   7  6

[image]

See layer 163, tool: 634332 - 634882
[Test_V_Skele_Uni_25x07_3days_gcode.zip](https://github.com/prusa3d/PrusaSlicer/files/5717570/Test_V_Skele_Uni_25x07_3days_gcode.zip)

[Test_V_Skele_Uni_25x07_3days.zip](https://github.com/prusa3d/PrusaSlicer/files/5717513/Test_V_Skele_Uni_25x07_3days.zip)

bubnikv, 2021-01-05:
I see the issue. Unfortunately as of now PrusaSlicer considers the whole multi-material slice a single object for travel path optimization, the travel path optimization does not "see" the smaller blue islands as separate entities. That is a deficiency, which should be fixed one day. Unfortunately it will take some time as the G-code generator is quite convoluted, also due to the "wipe into infill" and "wipe into another object" features.

bartrv, 2021-01-06:
Thank you for looking into this.

I figured it was more fundamental when I saw another post about similar behavior with sparse infills that “corrected” when the infill became more dense.

bubnikv, 2023-05-12:
Fixed in [2.6.0-alpha2](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha2)

closed 2023-05-12 by bubnikv

unrepentantgeek, 2023-10-21:
This issue is still(again?) present in 2.6.1 and 2.7.0-alpha1.  Project file attached with model designed to surface the issue.

OS: Ubuntu Linux 22.04
PrusaSlicer 2.7.0-alpha1 (also seen in 2.6.1)

[Path planner test case.zip](https://github.com/prusa3d/PrusaSlicer/files/13061437/Path.planner.test.case.zip)