prusa3d/PrusaSlicer #5513: internal error post_init() while not yet initialised with PrusaSlicer 2.3.0-rc1

SittingDuc, 2020-12-18:
### Version
PrusaSlicer-2.3.0-rc1+linux-x64-GTK3-202012172045.AppImage
PrusaSlicer-2.3.0-rc1+linux-x64-202012172035.AppImage

### Operating system type + version
Linux Ubuntu 18.04.5 LTS, kernel 4.15.0-124-generic #127-Ubuntu SMP Fri Nov 6 10:54:43 UTC 2020 x86_64

### 3D printer brand / version + firmware version (if known)
didn't get that far, but MK3S

### Behavior

Previously used PrusaSlicer 2.2.0-rc4 and PrusaSlicer 2.3.0-alpha1 without issue. Attempted to open the new PrusaSlicer 2.3.0-rc1 and got the splash screen with the orange egyptian lady and then a dialog box with an error 'Internal error: calling post_init() while not yet initialized' and returns to the console.
redirecting stdout and stderr, stdout was empty but stderr contained the following log.
[stderr.txt](https://github.com/prusa3d/PrusaSlicer/files/5718130/stderr.txt)

Attempt 2 using the nonGTK3 version produced a similar but different log file
[stderr_nogtk3.txt](https://github.com/prusa3d/PrusaSlicer/files/5718139/stderr_nogtk3.txt)

May be related to #5074 

Steps to reproduce:
* Download AppImage from github
* Stick in a directory
* Copy ~/.PrusaSlicer from 2.2.0 to ~/.config/PrusaSlicer for 2.3.0
* Attempt to run PrusaSlicer-2.3.0-rc1+linux-x64-GTK3-202012172045.AppImage or PrusaSlicer-2.3.0-rc1+linux-x64-202012172035.AppImage

Expected result: splash screen then a gui with a grid
Actual result: splash screen, dialog box with post_init internal error and back to the console.

Next I may try wiping .config/PrusaSlicer and see what that does... After that, I guess I have to compile from source and see what that does...

SittingDuc, 2020-12-18:
okay, erased both ~/.PrusaSlicer and ~/config/.PrusaSlicer and this time the AppImage worked and loaded the first-time-wizard. Selected my printer and filament and received the gui with the build plate - as expected.

Closed the program and reopened it, and received the gui with the build plate - as expected.

Conclusion: There must be a setting in ~/.PrusaSlicer that I copied to ~/.config/PrusaSlicer that made the software unhappy on load up.

I guess I should go diff the files and see if I can narrow down the culprit, so my little issue can have meaning ;)

SittingDuc, 2020-12-18:
`rm -f ~/.config/PrusaSlicer/printer/Original\ Prusa\ i3\ MK3S\ -\ Octopi.ini`

and it works. So that file is causing the error message.
After removing my IP Address and API Key, here is a copy of the relevant file. Why does this file cause a 'post_init' internal error?

[Original_Prusa_i3_MK3S__Octopi.ini.txt](https://github.com/prusa3d/PrusaSlicer/files/5718237/Original_Prusa_i3_MK3S__Octopi.ini.txt)

lukasmatena, 2020-12-19:
Thanks for the report. I can see the issue and will fix it tomorrow.

lukasmatena, 2021-01-04:
Should be fixed in [2.3.0-rc2](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.3.0-rc2). Closing.

closed 2021-01-04 by lukasmatena

SittingDuc, 2021-01-10:
I can confirm that I using PrusaSlicer 2.3 rc3, I can copy my PrusaSlicer 2.2 config over and start the program, and it opens without an error message. So Fixed, NowWorksForMe, etc.

Thank you!