prusa3d/PrusaSlicer #5514: Program Crash: NotificationManager.cpp, when exporting GCode

n8bot, 2020-12-18:
### Version
Slightly modified version of the master branch as at 2fa5828023c1bc9a7efa77b3da135decd284b418

### Operating system type + version
Win 7 64 bit

### Behavior
When modifying Start/End/Tool change GCode in Printer custom GCodes, then exporting the GCode after slicing, the program crashes. It does not crash every time, only intermittently but it has happened several times and always while doing the same thing. It does not seem to happen the first time GCode is exported, but after repeatedly making changes, slicing, exporting.

I happen to be running it from a compiled source so have some limited debugging info from VS. See below:

[image]

#### Project File (.3MF) where problem occurs
3mf in ZIP:
[IDEX Calibration.zip](https://github.com/prusa3d/PrusaSlicer/files/5718220/IDEX.Calibration.zip)

kocikdav, 2020-12-19:
Hello, we have found one missing init call in notifications. It might be reason of your crash. Please would you try 593c397b1915fe1784645e5f2898c42918823e6d or this build https://www.dropbox.com/s/g0a6bgk3em02miy/PrusaSlicer-2.3.0-rc1%2B13-win64-g593c397b1-202012190924.zip?dl=0 ? It would be great help if you tested this for us. Thanks.

n8bot, 2020-12-19:
Sure thing, I'll give it a try and attempt to reproduce the problem. Since it was intermittent, I don't know if I will be able to conclusively say if there is any difference but I will try my best to replicate my steps.

I didn't read the code very much, but it looks like this has to do with mouse-over events? It would make sense, I was switching between programs fast while exporting GCode, so my cursor might have floated over or off of a window at the wrong time? 

I'll let you know if I can reproduce the crash.

n8bot, 2020-12-19:
I repeatedly edited the custom G-code, sliced, then exported the gcode as I was before. I did this for about 5 minutes straight and there were no crashes. It seems silly for me to continue doing that over and over. It *feels* like things are fixed, because I didn't see a crash or even a stutter so for now let's call it fixed!

lukasmatena, 2020-12-19:
Lovely. We did not manage to reproduce it, let's hope we actually fixed the cause. It would be nearly impossible without the debugging info that you provided, so thank you very much.

Closing the issue, keep watching us before the final.

closed 2020-12-19 by lukasmatena

lukasmatena, 2020-12-22:
Fixed in [2.3.0-rc2](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.3.0-rc2) (593c397b1915fe1784645e5f2898c42918823e6d).