prusa3d/PrusaSlicer #5515: Feature Request: Disable "Wipe while retracting" on First Layer

Shootquinn, 2020-12-18:
### Version
_Version of PrusaSlicer used goes here_
2.2.0 & 2.3.0-rc1

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_
Windows 10 19042.685
### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_
Prusa Mk3 (Not S), Prusa Mini+ FW 4.2.1-final-2072 

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

Please add an option to disable wipe on the first layer. This is important for first layer quality on small details, including text. In the included .3mf, I have set first layer accelerations to 125 mm^s/2 to reduce chances of pulling up small parts of the first layer. When wipe while retracting is enabled, these careful print moves are followed by a sudden wipe, apparently to use the "default" acceleration value. To remedy this, I unchecked Wipe while retracting and had a successful print. However, Wipe while retracting is desirable for the Prusa Mini, which is why it is enabled by default. Note that the acceleration behavior is predictable and correct to achieve a good retraction, it is only desired to disable the wipe altogether in this case. 

Ref. issue #3493, the behavior described in that issue is not evident in my test case. Adding this feature may satisfy issue #3493 as well, by being able to control the behavior. 

Ref. issue #2507 comment thread for discussion of wipe effects on top layer. After reading this thread I commented, concluding: "While I envisioned a toggle for disabling wiping on the first layer, the fact that it also affects top layer quality may demand more flexibility. Aside from the top and bottom layer, wiping is mainly to prevent stringing and ooze. For the top and bottom layer, however. wiping contributes/detracts to the finish quality directly."

Thus it may be best to offer more precise control of what layers Wipe while retracting is enabled on. A checkbox for "Wipe on all layers" followed by a pair of spin boxes for "Wipe on xx bottom layers" and "Wipe on xx top layers" may be a solution. 

_Is this a new feature request?_
YES

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
Example 3mf attached for viewing layer 1 wipe behavior. 
[HelioxGuide_Wipe.zip](https://github.com/prusa3d/PrusaSlicer/files/5718362/HelioxGuide_Wipe.zip)

_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_
Screenshot from example .3mf included. 
[image]

Area5142, 2020-12-19:
This issue combined with the printing order on first layer issue #5212 often gives me problems when printing text in PETG.
The only solution I have found is to slow down printing speed to 40-50% on the physical printer on the first layer until solid infill begins. It is not enough to set first layer speed in PS as travel speed is unchanged - doing it on the printer will slow down all speed settings.

Shootquinn, 2020-12-19:
One tip for the gap fill: with version 2.20 (haven't verified that it still works on 2.3.0-rc1) you can disable gap fill by setting the gap fill speed to zero. In the main slic3r branch there is a checkbox for gap fill, but in PrusaSlicer it appears disabling it is done by setting the speed to zero. Also, in the past I have used a modifier to only do this to the first layer. Unfortunately, you can't do this with the wipe setting, so you would have to slice the object twice (with wipe disabled, and with it enabled) and copy/paste the first layer gcode to have a different setting on the first layer.

JFingerle, 2022-09-29:
I would really like to have this feature. Is there any update on this issue yet?

GeorgeBekh, 2023-02-12:
Check out my python scripts which removes wipes from the first layer. It works using comments in gcode generated by prusa slicer
  
https://gist.github.com/GeorgeBekh/bf7af0bd61430426e042c63f18560832

somethingelseentirely, 2025-11-23:
Shouldn't this just be added to the modifier setting, so that it can be disabled for certain layer ranges?

closed 2026-09-29 by prusaslicer-bot[bot]