prusa3d/PrusaSlicer #5517: "Unsupported Selection" error when copy pasting objects with modifiers, sometimes full crash to desktop

tensheapz, 2020-12-19:
### Version
PrusaSlicer-2.3.0-rc1+win64

### Operating system type + version
Windows 10 x64

### 3D printer brand / version + firmware version (if known)
Prusa MK3S

### Behavior

Popup error occurs when trying to duplicate multiple objects at once where the objects have printing modifiers. This also can cause a full crash to desktop although this is not reliably reproducible.

**Steps to reproduce**
- Create Simple Box - Right Click > Add Shape > Box
- Add any modifier (such as infill) to the shape - Right Click > Infill
- Duplicate the Box - Select the box, Ctrl+C, Ctrl+V
- In the second box's infill modifier, set it to a different value, such as 26%. This seems to be important that they have different modifiers
- Select both boxes (via Ctrl+Click) and duplicate them both simultaneously (via Ctrl+C, Ctrl+V)
**Actual Results**
-- Crash to desktop occurs
-- Unsupported selection popup occurs a couple of times before succeeding in the duplication

**Expected results**
Program does not crash, no error popup occurs, and the duplication happens as intended.

[image]

_Is this a new feature request?_
no

lukasmatena, 2020-12-19:
Thanks. The bug is already present in 2.2.0. When I tested, the error popup showed every time (for each object), objects were copied correctly, I never saw it crash (not saying that you couldn't have). We will fix this, but not for 2.3 final release.

lukasmatena, 2020-12-22:
The hard crash should no longer happen in [2.3.0-rc2](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.3.0-rc2) after 6c39905. The rest will be fixed later.

bubnikv, 2021-06-03:
Fixed in PrusaSlicer 2.4.0-alpha1.

closed 2021-06-03 by bubnikv