prusa3d/PrusaSlicer #5518: Cannot start Prusa Slicer on Dindows

jan-b45, 2020-12-19:
### Version
2.2.9.1
also tried 2.3.0 alpha 2
### Operating system type + version
Windows 10 20H2

### 3D printer brand / version + firmware version (if known)
MK2,5S + MMU2

### Behavior
I am not able to start the Prusa Slicer. After duble click on the icon, it shows nothing. I have only error in event log that says application error 0xc000041d nothing more. I tried uninstall Prusa slicer, delete all files from AppData/Roaming, Program Files, from drivers and from registry using Revouninstaller and install again, but without any success. I tried even aplha version 2.3.0 aplha2, but same result. 
Computer does not have any changes or updates in last 2 weeks and application stops work from day to day.

#### Project File (.3MF) where problem occurs

fe60, 2020-12-21:
I was not able to reproduce the problem with 2.3.0-beta1 and 2.3.0-rc1.
Both were working fine on Windows 10 2004 and still do on 20H2 (just did the update).

Does the issue occur when Windows is started in safe-mode [1]?
Is there a virus scanner or a similar software, that could cause the problem?

[1] https://support.microsoft.com/en-us/windows/start-your-pc-in-safe-mode-in-windows-10-92c27cff-db89-8644-1ce4-b3e5e56fe234

jan-b45, 2020-12-21:
PC is not inficated with virus, I made severa scans with Malwarebytes, Bitdefender and Winows Defender. Slicer works perfectly, but one day just stop, without any system or software changes. Computer was in sleep mode and after wake up, Slicer does not work. Tried reboot, uninstall, clear registry, app folders but nothing helps. 
Slicer works in safe mode

fe60, 2020-12-22:
I don't suppose your PC to be infected by a virus. I rather suspect one of the virus scanners to cause the problem.
Did you try to run PrusaSlicer without being scanned (deactivate scanner or add folder with PrusaSlicer to exclusion list)?

I'm not an export for crash analysis, but I guess more details will help.
Just see https://github.com/prusa3d/PrusaSlicer/issues/4316#issuecomment-634769532 asking for detailed log

jan-b45, 2020-12-22:
Unfortunately, it did nothing, when I run it with loglevel parameter. I tried do disable Windows Defender and uninstall Malwarebytes, but nothing change.
[image]

BrerDane, 2021-04-03:
I'm having the same trouble after Windows updated build 19042.906 version 20H2. I forgot to extend my paused updates. Reset security settings (e.g., enabled Cortana) so not sure what was reset on my ASUS. Dell is still on build 19041.867 version 2004 and it works fine.

Shows up in task manager for a few seconds if I run as admin then disappears.

jan-b45, 2021-04-06:
In my case was issue with Toshiba docking station driver. It does not matter, if I connect my NTB to docking station or not, but only with installed driver, I was not able to open Slicer. Only uninstall driver a new install works. Unfortunately Toshiba released latest driver 2 years ago for my docking station.

vexorg3000, 2021-06-11:
I am experiencing the same issue. 
The prusa folder is excluded from scanning.
[image]

also please see the error msg from the windows event log

Log Name:      Application
Source:        Application Error
Date:          11/06/2021 3:13:50 PM
Event ID:      1000
Task Category: (100)
Level:         Error
Keywords:      Classic
User:          N/A
Computer:      XXXX.blackmores.com.au
Description:
Faulting application name: prusa-slicer.exe, version: 2.3.1.0, time stamp: 0x60799a27
Faulting module name: unknown, version: 0.0.0.0, time stamp: 0x00000000
Exception code: 0xc000041d
Fault offset: 0x0000000000000000
Faulting process ID: 0x5e88
Faulting application start time: 0x01d75e808f3c77e6
Faulting application path: C:\Program Files\Prusa3D\PrusaSlicer\prusa-slicer.exe
Faulting module path: unknown
Report ID: f97bff9b-43d8-4330-b999-da97c4e71268
Faulting package full name: 
Faulting package-relative application ID: 
Event Xml:
<Event xmlns="http://schemas.microsoft.com/win/2004/08/events/event">
  <System>
    <Provider Name="Application Error" />
    <EventID Qualifiers="0">1000</EventID>
    <Version>0</Version>
    <Level>2</Level>
    <Task>100</Task>
    <Opcode>0</Opcode>
    <Keywords>0x80000000000000</Keywords>
    <TimeCreated SystemTime="2021-06-11T05:13:50.3668438Z" />
    <EventRecordID>8196</EventRecordID>
    <Correlation />
    <Execution ProcessID="0" ThreadID="0" />
    <Channel>Application</Channel>
    <Computer>XXXX.blackmores.com.au</Computer>
    <Security />
  </System>
  <EventData>
    <Data>prusa-slicer.exe</Data>
    <Data>2.3.1.0</Data>
    <Data>60799a27</Data>
    <Data>unknown</Data>
    <Data>0.0.0.0</Data>
    <Data>00000000</Data>
    <Data>c000041d</Data>
    <Data>0000000000000000</Data>
    <Data>5e88</Data>
    <Data>01d75e808f3c77e6</Data>
    <Data>C:\Program Files\Prusa3D\PrusaSlicer\prusa-slicer.exe</Data>
    <Data>unknown</Data>
    <Data>f97bff9b-43d8-4330-b999-da97c4e71268</Data>
    <Data>
    </Data>
    <Data>
    </Data>
  </EventData>
</Event>

neaxi, 2021-08-10:
Had the same `0xc000041d` issue today. I've removed all previous instances of PrusaSlicer including all related folders in `%APPDATA%` and downloaded the newset version from the releases page, but it still failed to launch.

No output for `prusa-slicer-console.exe --loglevel=5` (the same as jan-b45)
```
C:\Users\_\Desktop\PrusaSlicer-2.3.3+win64-202107161027>prusa-slicer-console.exe --loglevel=5

C:\Users\_\Desktop\PrusaSlicer-2.3.3+win64-202107161027>
```

**Windows event logs**:
```
Faulting application name: prusa-slicer.exe, version: 2.3.3.0, time stamp: 0x60f1604d
Faulting module name: unknown, verze: 0.0.0.0, time stamp: 0x00000000
Exception code: 0xc000041d
Fault offset: 0x0000000000000000
Faulting process ID: 0x36e4
Faulting application start time: 0x01d78dedb6d6e47f
Faulting application path: C:\Users\_\Desktop\PrusaSlicer-2.3.3+win64-202107161027\prusa-slicer.exe
Faulting module path: unknown
Report ID: 6ebe3f25-fd10-454d-b2cc-e4d50e18f865
Faulting package full name:
Faulting package-relative application ID:
```

**Following fix helped**:
https://forum.prusaprinters.org/forum/prusaslicer/prusa-slicer-crashing/paged/2/
a) temp / test: `prusa-slicer-console.exe --sw-renderer --loglevel=5`
b) permanent: _"switch permanently to the software renderer by copying the MESA/OpenGL.dll one level up next to prusa-slicer.exe."_

**Confirmed by following**: 
If OpenGL.dll is in the same folder as prusa-slicer.exe, the application launches correctly. 
When removed/moved back to the mesa folder, the startup issue occurs again.

I'm facing similar issue with Stellarium application. Have to pick correct renderer otherwise the application won't launch.

**Suspected cause - drivers**:
Missing `igdgmm64.dll` (Intel HD Graphics driver). The process attempted to search all `%PATH%` locations for it, haven't found any results and then crashed (see screenshot below). Not sure why the file is missing as the drivers are up to date.
(Same as jan-b45 I'm using docking station, which presents itself as a graphical adapter and works via DisplayLink)

[image]

I've managed to find `igdgmm64.dll` in `C:\Windows\System32\DriverStore\FileRepository\iigd_dch.inf_amd64_a086f01cc7be643a`.
When placed in `%PATH%` the PrusaSlicer finds the driver and launches, but results in following error which informs the user he needs to use the software renderer.

[image]

bubnikv, 2021-08-10:
@neaxi I am sorry, but your graphics driver installation seems to be messed up. You may try to reinstall your graphics driver.

neaxi, 2021-08-11:
> @neaxi I am sorry, but your graphics driver installation seems to be messed up. You may try to reinstall your graphics driver.

I agree the issue **in my case** are graphics and their drivers. (although both Intel and Nvidia has July 2021 version installed)

My point was that when GPU driver (i.e. `igdgmm64.dll`) is not found I would expect some output informing user about what went wrong (same style as when provided outdated driver without OpenGL 2.0?). But as you can see even lanching via `prusa-slicer-console.exe --loglevel=5` provides no information or hint on what may be wrong.

bubnikv, 2021-08-11:
I don't think we can catch such an error. These DLLs are loaded
implicitly by other DLLs, the poor windows application has no way of
detecting it.

closed 2026-09-29 by prusaslicer-bot[bot]