prusa3d/PrusaSlicer #5519: Slicer produces paths not always conforming to geometry

protonenfalter, 2020-12-19:
### Version
_Version of PrusaSlicer used goes here_

Version 2.3.0-rc1+linux-x64-GTK3

### Operating system type + version
Ubuntu 20.04
### 3D printer brand / version + firmware version (if known)
Prusa i3MK3

### Behavior
Slicing a specific geometry ('NoOverhangs.stl') without overhangs produces G-code which in the preview of slic3r shows pronounced overhangs. See attached screenshot.
[image]

_Is this a new feature request?_
No.

#### Project File (.3MF) where problem occurs
'Overhang.3mf'
[overhang.zip](https://github.com/prusa3d/PrusaSlicer/files/5719454/overhang.zip)

rtyr, 2020-12-19:
Disable "Detect thin walls" option.

protonenfalter, 2020-12-19:
Ok, that solves the issue. Thank you!

closed 2020-12-19 by protonenfalter