prusa3d/PrusaSlicer #5520: <feature proposal> Ensure minimal travel in "free air" when "Avoid crossing perimeters" is active

MB3Y, 2020-12-19:
### Version
2.3.0 - alpha3+win64
### Operating system type + version
Win10
### 3D printer brand / version + firmware version (if known)
MK3S+MMU2S
### Behavior
There are still long travels in "free air" when avoid crossing perimeters is active in case e.g. there is support to be printed / multi regions / ...
These travels are now often lines from A to B (close to the direct path from A to B) avoiding correctly the perimeters, ... on the OUTSIDE of the object area.
To the strict meaning of the terminology and name of the feature that is fine. When interpreting that to the intention behind the feature it is less obivous/acceptable. The intention of the avoid perimeters crossing I see as "avoid both oozing, stringing as well as visisble artefacts on the print perimeters that are caused by a nozzle that moved outside the parts boundaries/perimeters." This is in line with the tooltip/helptext when hoovering over the current tickbox for this feature.

* _Expected Results_
Stay within the objects boundaries as long as possible, find the closest point within the object area where point A belongs to, and jump only from there to B. This would minimize the length of the "free air" travel and minimize the possibility for leakages, strings or other artefacts becoming visible outside the object.
_The same holds when jumping back into the object._
As an alternative possibility one could also not jump from the closest point, but from a nearby/closest corner - potentially hidding any artefacts even more.

Evidently the risk in implementing such feature is to end up with a potentially extra long travel move (e.g. assume one has to jump to the circumference of a spiralling snake from the center of that snake...)
I recon a slave parameter retricting the max travel distance or time is either already present or could be added?

* Screenshots from *PrusaSlicer* with an indication of the intention added below
_Is this a new feature request?_
Yes
[image]
[image]

MB3Y, 2020-12-19:
Sorry about the spiral example reference ... only now reading the release notes, where you also indicate tons of adjustments on exactly this feature and a spiral example... no offense! The feature DOES behave much better than before already...
PS: the release notes mention even a parameter "max detour length", but I don't find that accessible in the GUI?

ntmaster26, 2020-12-19:
Unsubscribe

lukasmatena, 2020-12-19:
> the release notes mention even a parameter "max detour length", but I don't find that accessible in the GUI?

The Avoid crossing perimeters was completely rewritten in 2.3.0-**beta3**, you seem to be testing 2.3.0-alpha3, which most likely makes your feature request obsolete. The most recent version to test at this point is [2.3.0-rc1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.3.0-rc1).

We will gladly hear your opinion about the new implementation.

MB3Y, 2020-12-21:
The resulting behavior in rc1 is basically the same. The jumps from the objects infill to the support outside the object happens directly, without optimization of travel path within the object first. In the below example the travel even crosses multiple times such boundary.
I can only assume that in case multiple areas
[image]

I also made this quick test object - that has 3 towers on a common base - so crossing perimeters is mandatory. This ticket proposes to optimize the travel paths in "free air" such that they either represent the shortest distance or the shortest distance reducing surface artefacts (what I presume is jumping to corners if possible)
Below picture demonstrates the possibilities, 1. being the simplest (in description), 2. tries to add optimization of surface artefacts and 3. considers "free to air" time too
[image]

closed 2026-09-29 by prusaslicer-bot[bot]