prusa3d/PrusaSlicer #5521: Build Plate model changes at random

Area5142, 2020-12-19:
### Version
PrusaSlicer version 2.3.0 rc1 g2fa582802 and RC1

### Operating system type + version
Linux Mint 19.3 x64

### 3D printer brand / version + firmware version (if known)
Not relevant - same for all printers.

### Behavior
* _Describe the problem_
Starting PS with one of the Prusa Printers (MK3S, MMU2, Mini and SL1) sometimes shows a model of the physical build plate - at other times PS shows a generic build plate.
It appears to be random when this happens. It can happen when starting PS and it can happen switching from one printer to another or just changing printer settings.

* _Steps needed to reproduce the problem_
I can't provide a water proof method to provoke this error, but it happens often and to my observations at random. No model needs to be loaded for this to happen.
Changing printer model will sometimes provoke the error or just closing and reopening the slicer.

* _Expected Results_
Always open with build plate model for the selected printer.

* _Actual Results_
Sometimes opens with generic build plate, see screen shots.

* _Screenshots from __*PrusaSlicer*__ preview are preferred_

Correctly shown build plate:
[image]

Generic build plate shown:
[image]

I noticed that the printer profile is different on the two screen shots but that is not important as this happens with the same profiles and different physical printers too.

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
Not needed.

lukasmatena, 2020-12-20:
I can see from the screenshots that there are at least two Printer profiles, "Original Prusa SL1" and "Original Prusa SL1 - JOH". What is the difference between them? Are you sure that both have their print bed model/texture defined?

Area5142, 2020-12-20:
The changed profile on the first screen shot was a standard profile from 2.2.0 with _begin_ and _end_ g-code changed to stop nagging me with newer firmware and lift the extruder to min 100 mm after a print is finished. No changes to printer bed image/model.

The missing/changed print bed also happen with the latest profiles:

Here the SL1 standard profile loaded correct with empty print bed:
[image]

Here the same profile in same session after changing standard Mini+/MK3+/MMU2 a couple of times:
[image]

It seems to be a random error, sometimes the print bed is shown wrong when PS is just started other times it happens when I just change profile between the different printers: SL1/MK3S+/Mini/MK3S+MMU2S printer.

lukasmatena, 2020-12-20:
Thanks for the clarification. I did not reproduce it on my system, though.
Did you have these issues with 2.2.0 release?

Area5142, 2020-12-21:
Have not seen this problem in version 2.2.0.

Just tried to provoke the error in version 2.2.0, but can't get it to fail.

closed 2026-09-29 by prusaslicer-bot[bot]