prusa3d/PrusaSlicer #5522: missing hole in sliced output

mixicz, 2020-12-19:
### Version
2.2.0+linux-x64

### Operating system type + version
Debian GNU/Linux 11 bullseye (testing)

### 3D printer brand / version + firmware version (if known)
Prusa i3 MK3S (not modified), FW 3.9.1.3518

### Behavior
There is missing hole in sliced output – imported object is rendered correctly, but after slicing, one hole is filled, see screenshots below.
Print settings does not seem to matter (I have tried different presets, various filaments and even changed printers). I use mostly stock settings presets, for example:
* printer: Original Prusa i3 MK3S & MK3S+
* Filament: Prusament PLA
* Print settings: 0.20mm QUALITY

[image]
[image]

#### Project File (.3MF) where problem occurs
[dock_mon_holder-error.zip](https://github.com/prusa3d/PrusaSlicer/files/5719921/dock_mon_holder-error.zip)

rtyr, 2020-12-19:
It is most likely caused by errors in the model.

Fixed model:
[dock_mon_holder-error_repaired.zip](https://github.com/prusa3d/PrusaSlicer/files/5720213/dock_mon_holder-error_repaired.zip)

mixicz, 2020-12-20:
Slicer reports that model is OK and is rendered correctly, so I consider this slicing error. I was able to work around this by removing outer chamfer from the affected part (it might alter model in another ways as well though).

rtyr, 2020-12-20:
It slices fine with repaired model.

closed 2026-09-29 by prusaslicer-bot[bot]