prusa3d/PrusaSlicer #5523: Feature request:

GeekDad63, 2020-12-19:
### Version
2.3 rc1

### Operating system type + version
Mac OS 11 Big Sur

### 3D printer brand / version + firmware version (if known)
Prusa MK3

_Is this a new feature request?_
I wish that the names of the objects in the sidebar could dynamically collapse so it is only as big as the amount of items. This way the other stuff below it would be visible without scrolling down.

lukasmatena, 2020-12-19:
Thanks. Would the discussion in https://github.com/prusa3d/PrusaSlicer/issues/5162#issuecomment-748373571 cover your suggestion? If so, can we close this issue and only leave the other one open?

GeekDad63, 2020-12-19:
I believe that #5162 (comment) discussion is that the names area is too small but my issue is that the object names area is bigger than it needs to be. On smaller laptop screens this takes up real estate.
[image]

lukasmatena, 2020-12-19:
The #5162 was indeed about the area being too small (in versions before 2.3.0-rc1). We increased the minimum for rc1 and someone else already complained (#5462) that it is too large. We have already made it smaller again and promised to do the dynamic sizing for the 2.4 release (which should fix it for good and for everyone). The size in current (unreleased) master is almost the same size as it was before the rc1. That's why I'd like to keep #5162 open. It is also the oldest of the issues so it contains this history.

agrestic1, 2020-12-20:
Yes after updating from 2.2 to 2.3rc1, suddenly the right area on my (common) 1920x1080 resolution is slightly too big so I need to scroll down to see just three lines with cost an print time

closed 2026-09-29 by prusaslicer-bot[bot]