prusa3d/PrusaSlicer #5526: bug with certain models

shikyojojo, 2020-12-20:
### Version
version 2.3 rc1

### Operating system type + version
 windows 10

### 3D printer brand / version + firmware version (if known)
stock mk3s and mini

### Behavior
some models/stls brought in say, object with no extrusions on first layer on stock profiles but if I use a very old profile the issue does not exist.   
  
[image]

[Body1.zip.zip](https://github.com/prusa3d/PrusaSlicer/files/5720370/Body1.zip.zip)

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

lukasmatena, 2020-12-20:
How would you print the model if the error wasn't there?

MunkeyBalls, 2021-02-09:
I ran into the same problem with 0.8 nozzle and vase mode.
The error went away after increasing the first layer height from 0.2 to 0.3, this profile was working before updating.

Running prusaslicer 2.3.0-rc2

closed 2026-09-29 by prusaslicer-bot[bot]