prusa3d/PrusaSlicer #5527: Post processing script gets wrong filename when using direct uploading to printer functionality

tomvandeneede, 2020-12-20:
### Version
_2.3.0 RC1+_

### Operating system type + version
MAC OS11.1
Prusa MK3S 3.9 - Octoprint

### Behavior
when directly uploading the file to the Octoprint server when using a post processing script, the post processing script seems to get an invalid name and cannot find the file.  Afterwards the uploading happens just normally
Sometimes it does work, but most of the time it fails..

This graphic shows on top what is in the folder (2 files from a crashed earlier slices), the bottom shows the script argument that is passed.  As you can see the file is not in the location it is expected to be in.   The screenshot was taken at the moment of execution of the script so the file should have remained here during that time

After processing the printer seems to upload the code just fine to the printer (though  not processed) (this is an error on the post processing script side... need to exit with proper error code when this happens

[image]

_Is this a new feature request?_  No

Happens with all models randomly, not model related

lukasmatena, 2020-12-20:
Do you also have this issue with 2.2.0? When do you think it started appearing? I do not see anything bad on Linux.

tomvandeneede, 2020-12-20:
Hi Lukas,

No 2.2 seems to be working ok.

I tried 2.3 Beta3 and RC1 and both with sometimes and sometimes they don’t

from the source code the only logical explanation would be copy_file returning a success in error…though it would not explain why the upload would still work unless this is using a different source for the path to upload

lukasmatena, 2020-12-20:
@tomvandeneede Thanks. We did not manage to reproduce it and to be honest, I'm not sure what can be wrong. Are you able to test our released 2.3 versions to see when it broke? Or was it already broken in 2.3.0-alpha1? Can you reproduce it with any script (including e.g. a simple sh script that just writes something into the file)?

When you run slicer with `--loglevel=5` parameter, it should log quite extensively, including the filenames. That might point us somewhere. What does the log say in the case when your script fails and the upload continues fine?

tomvandeneede, 2020-12-20:
will do that

as said I looked at the code as well and could not find anything particular 

will see if the extra logging yields some insight

tomvandeneede, 2020-12-20:
Lukas,

managed to solve it.   The open command seems to work asynchronously and that caused the sending to happen before the file opened
sorry for this… 

you can close the issue

lukasmatena, 2020-12-20:
Ok, so if I get it right, the problem is in your script, PrusaSlicer passes the filename correctly and the file is where it is suppose to be, right? That would be good news (for us :-)). Closing the issue then.

closed 2020-12-20 by lukasmatena

tomvandeneede, 2020-12-20:
yep apparently open -a <appname> worked asynchronously and did not wait until the processing was finished
not sure why this did work on PS2.2 then but maybe 2.3 has improved speed causing this to become an issue

Sorry for waisting your time, but had major issues with P2PP due to the recent OS11.1 upgrade and had due to the mix of working and not working in PS22 PS23 I assumed it was on your end

regards

lukasmatena, 2020-12-20:
No worries.