prusa3d/PrusaSlicer #5530: Prus-Slicer creates a GCODE wihich stops extruder after command g1 e-x F3600

bowie130459, 2020-12-20:
### Version
Version 2.2.0+

### Operating system type + version
Mac OSX 10.15.7

### 3D printer brand / version + firmware version (if known)
Ender 3d Pro Marlin 2.0.7
Titan Air hotend

### Behavior
A lot of search and analyse has beend done from my side and at least I found the following 
The extruder stops after the inital lines caused by.
G01 E-5 F3600

This code sniplet could be found several times with different e- values but always the same F3600 value.
I changed the first occurance of 3600 to 2400 and it works till the next occurance of F3600

A change of F3600 threw the whole Gcode to F2400 and the whole file could be processed.

This is no new Feature request only where is the setting of this value 3600 which seems to lead to an error on 
Ender 3d Pro machines with Marlin 2.0.7 and TITAN Aero Hotend.

Where is the standard Value to set this value

Thanks for your assistance.
Willi

[Projectfiles Cura and Prusa.zip](https://github.com/prusa3d/PrusaSlicer/files/5720770/Projectfiles.Cura.and.Prusa.zip)

#### Project File (.3MF) where problem occurs
[GCodefiles-Cura and Prusa.zip](https://github.com/prusa3d/PrusaSlicer/files/5720772/GCodefiles-Cura.and.Prusa.zip)

rtyr, 2020-12-20:
F3600 is your retraction speed (60mm/s). You can change it in the printer settings.

bowie130459, 2020-12-20:
> F3600 is your retraction speed (60mm/s). You can change it in the printer settings.

Hey this is the solution, but I wonder after changing to 30 mm/s F was set to 1800, which is quit ok. but changing back to 60 only brought back 2100mm/s. I think there was a hickup when 2.2.0+ was installed.

bowie130459, 2020-12-20:
It works

closed 2020-12-20 by bowie130459