prusa3d/PrusaSlicer #5531: Problem with displaying tabs

Denethor85, 2020-12-20:
### Version
Versione 2.3.0-rc1+win64

### Behavior
In the old version of Prusa Slicer 2.2.0, in the "printer settings" section, if you selected for example the "extruder 1" tab and then switched from one printer profile to another, the software continued to display the same tab. 
This was convenient for comparing the same tabs of 2 different profiles. 
Unfortunately the same thing does not happen in the new version and when changing profile always returns to the first "general" tab.

lukasmatena, 2020-12-22:
Should be fixed in [2.3.0-rc2](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.3.0-rc2) (738c115ec11d2e61fc439deaecd9fee4e98e0272). Please retest and let us know if it isn't.
Thanks for the report. Closing.

closed 2020-12-22 by lukasmatena

Denethor85, 2020-12-24:
In new Prusa Slicer 2.3.0 RC2 the tab displaying works fine again.
But there is still one thing that if I remember correctly it works differently from PrusaSlicer 2.2 or that would be a good thing anyway:
the possibility, once a printer profile has been selected, to jump from one to the other with the keyboard arrows.
If I remember correctly this happened in the old version and once a profile was selected it remained highlighted and thanks to the arrows you could scroll through the various profiles

YuSanka, 2020-12-29:
> the possibility, once a printer profile has been selected, to jump from one to the other with the keyboard arrows.
> If I remember correctly this happened in the old version and once a profile was selected it remained highlighted and thanks to the arrows you could scroll through the various profiles

But nothing is changed in its behaviour. Can You be more specific please. What exactly doesn't work for you and what is your operating system type + version

Denethor85, 2020-12-29:
Sorry if I don't know English well and I use google translate.

In version 2.2 when you selected a profile and moved with the arrows the profile remained highlighted, so you could continue to move with the arrows.

In version 2.3 when you select a profile then you move to another profile with the arrows, this will no longer be highlighted and therefore you will not be able to scroll with the arrows again.

I am attaching 2 videos to highlight the problem.

Denethor85, 2020-12-29:
https://user-images.githubusercontent.com/38705740/103317532-ba9f8e00-4a2b-11eb-8784-938491ba50f8.mp4

https://user-images.githubusercontent.com/38705740/103317533-bbd0bb00-4a2b-11eb-9cce-f4ea6b57caa6.mp4