prusa3d/PrusaSlicer #5533: Support in holes always generates little towers on both sides

fleaz, 2020-12-20:
### Version
2.2.0

### Operating system type + version
Arch Linux

### 3D printer brand / version + firmware version (if known)
Crealty Ender-3 (But unimportant to this issue)

### Behavior
Hey,
I was "Team Cura" the last few years but because I will get an i3MK3S for Christmas I wanted to switch back to PrusaSlicer because I think due to the premade and optimized profiles for all the Prusa machines, this would produce way nicer prints without any tinkering on my side.
After playing around a bit with PrusaSlicer, I encountered a behavior that kind of annoys me and already did years back: When generating support for a hole, PS always generates these little support towers on both sides. I made a simple block with a hole in the middle to demonstrate the difference between PrusaSlicer and e.g. Cura.
Especially in taller objects where the holes are on the top, this behavior generates a lot of unnecessary support structure.

Result when sliced in PS: (This is my Ender3 Profile but e.g. the official i3mk3s Profile generates the same "problem")
[image]

Same model sliced in Cura:
[image]

I'm currently not sure if this has to be seen as a bug/enhancement, so if this is intended behavior and can be in some kind deactivated with some settings that I haven't found until now, please let me now :)

Best regards,
Felix

#### Testobject
The STL from the screenshots
[testobject.zip](https://github.com/prusa3d/PrusaSlicer/files/5720936/testobject.zip)

Kachidoki2807, 2020-12-23:
Hello @fleaz,
The STL is not enough, and for this case, it is highly dependent on how you configured your supports parameters. For instance, this is how I got them:
[image]

bubnikv, 2021-09-22:
Solved with snug supports in PrusaSlicer 2.4.0-alpha1.

closed 2021-09-22 by bubnikv