prusa3d/PrusaSlicer #5534: macOS Prusaslicer 2.3.0-rc1 missing Creality Profiles

petercrepeau, 2020-12-20:
### Version
Version 2.3.0-rc1+

### Operating system type + version
macOS Big Sur 11.1

### 3D printer brand / version + firmware version (if known)
Creality CR-10s Pro running TH3D 2.13 firmware based on Marlin 2.0

### Behavior
All Creality profiles except fro Ender 3 are not present:
[image]

From Windows version of Prusaslicer 2.3.0-rc1+win64 I see the following:
[image]
[image]
[image]

_Is this a new feature request?_
No

rtyr, 2020-12-20:
You are likely using old configuration bundle.

petercrepeau, 2020-12-23:
Yes, that fixed the issue.  I thought I had done that already.

closed 2020-12-23 by petercrepeau