prusa3d/PrusaSlicer #5535: PrusaSlicer 2.3.0-rc1 should list "Prusa" as a firmware type

jcwren, 2020-12-20:
When configuring a new printer profile, it's not obvious what firmware type selection should be made when using Prusa stock firmware on a Prusa printer. Prusa should be added as a list entry, or at the very least put in parenthesis next to the correct entry.

[image]

rtyr, 2020-12-20:
> When configuring a new printer profile, it's not obvious what firmware type selection should be made when using Prusa stock firmware on a Prusa printer. Prusa should be added as a list entry, or at the very least put in parenthesis next to the correct entry.
> 
I don't see any reason to create custom profile for Prusa printer this way when you can simply select proper model on Prusa FFF page or create custom profile based on stock profile. 
 
> Probably should be a second bug report, but the MK3S+ is not listed as a printer type on the 'Prusa FFF' page.
> 
Are you sure?
[image]

lukasmatena, 2020-12-20:
Ready-made profiles for Prusa printers are much easier to obtain from the Prusa FFF page. Anyone bold enough to define a custom printer profile from scratch will face much harder issues than finding out that Prusa printers use slightly modified Marlin.

MK3S+ profile should be available (see below). Make sure that you have the latest configuration update installed (Configuration->Check for updates).
[image]

jcwren, 2020-12-20:
Yes, I'm pretty sure that my version of 2.3.0-rc1 does NOT show a MK3S+. Note the version in the title bar, and the lack of MK3S+ in the MK3 Family window.

I will also say that it is not terribly clear that I should NOT define a custom profile name in normal use. When you're just going through the setup, the implication is that you'd name this one thing for your first printer, and something else for a second printer. Perhaps that should be clarified to indicate it should only be selected if you're using a non-Prusa standard printer?

[image]

RedS-DEV, 2020-12-21:
> Yes, I'm pretty sure that my version of 2.3.0-rc1 does NOT show a MK3S+. Note the version in the title bar, and the lack of MK3S+ in the MK3 Family window. Update: This was a fresh install of -RC1. After doing a check for updates, it says there is one available. Perhaps it should do this initially when starting a configuration update? Update 2: I had to restart PrusaSlicer before the MK3S+ showed up in the MK3 Family window.
> 
> I will also say that it is not terribly clear that I should NOT define a custom profile name in normal use. When you're just going through the setup, the implication is that you'd name this one thing for your first printer, and something else for a second printer. Perhaps that should be clarified to indicate it should only be selected if you're using a non-Prusa standard printer?
> 
> [image]

I think you should use MK3S profile

closed 2024-02-18 by lukasmatena