prusa3d/PrusaSlicer #5537: G-code Viewer segfaults on Linux when gcode file is given as an commandline argument

hadrava, 2020-12-20:
### Version
PrusaSlicer G-code
Version 2.3.0-rc1+linux-x64-GTK3
Version 2.3.0-rc1+linux-x64
(Same issue on both versions)

### Operating system type + version
Debian Buster (10.7) x86_64 GNU/Linux

### Behavior

If i run G-code Viewer with .gcode file given as an argument, the program crashes with Segmentation fault:
```
$ PrusaSlicer-2.3.0-rc1+linux-x64-202012172035/bin/prusa-gcodeviewer 3DBenchy_0.2mm_PETG_MK3S_1h27m.gcode 
```
This happen on my computer with basically any .gcode file which is small enough. It can be also triggered from PrusaSlicer File --> G-code preview (as this spawns the 

Viewer with .gcode file given as an argument).

#### Observations from gdb

Running the G-code Viewer in gdb and printing backtrace points towards `Slic3r::GUI::GCodeViewer::load_toolpaths` function.
```
$ gdb PrusaSlicer-2.3.0-rc1+linux-x64-202012172035/bin/prusa-gcodeviewer 
GNU gdb (Debian 8.2.1-2+b3) 8.2.1
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from PrusaSlicer-2.3.0-rc1+linux-x64-202012172035/bin/prusa-gcodeviewer...done.
(gdb) r 3DBenchy_0.2mm_PETG_MK3S_1h27m.gcode --loglevel 5
Starting program: /tmp/PrusaSlicer-2.3.0-rc1+linux-x64-202012172035/bin/prusa-gcodeviewer 3DBenchy_0.2mm_PETG_MK3S_1h27m.gcode --loglevel 5
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[2020-12-20 17:24:06.688489] [0x00007ffff4e905c0] [trace]   System language detected (user locales and such): en_US
[2020-12-20 17:24:06.689213] [0x00007ffff4e905c0] [trace]   Best translation language detected (may be different from user locales): en_US
[2020-12-20 17:24:06.689242] [0x00007ffff4e905c0] [trace]   Switching wxLocales to en_US
[Detaching after fork from child process 30605]
[Detaching after fork from child process 30607]
[Detaching after fork from child process 30609]
[Detaching after fork from child process 30611]
[Detaching after fork from child process 30613]
[Detaching after fork from child process 30615]
[Detaching after fork from child process 30617]
[Detaching after fork from child process 30619]

(prusa-gcodeviewer:30601): GLib-GObject-WARNING **: 17:24:06.998: invalid (NULL) pointer instance

(prusa-gcodeviewer:30601): GLib-GObject-CRITICAL **: 17:24:06.998: g_signal_handlers_block_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(prusa-gcodeviewer:30601): GLib-GObject-WARNING **: 17:24:06.998: invalid (NULL) pointer instance

(prusa-gcodeviewer:30601): GLib-GObject-CRITICAL **: 17:24:06.998: g_signal_handlers_unblock_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(prusa-gcodeviewer:30601): GLib-GObject-WARNING **: 17:24:06.998: invalid (NULL) pointer instance

(prusa-gcodeviewer:30601): GLib-GObject-CRITICAL **: 17:24:06.998: g_signal_handlers_block_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(prusa-gcodeviewer:30601): GLib-GObject-WARNING **: 17:24:06.998: invalid (NULL) pointer instance

[... 184 lines cut ...]
[New Thread 0x7ffff2974700 (LWP 31779)]
[2020-12-20 17:44:18.465731] [0x00007ffff4e905c0] [info]    Undo / Redo snapshot taken: New Project, Undo / Redo stack memory: 0MB Resident memory: 60MB; Shared memory: 40MB; Private memory: 21MB; Peak memory usage: 145MB
[New Thread 0x7ffff2124700 (LWP 31780)]
[2020-12-20 17:44:18.468074] [0x00007ffff2124700] [error]   Not primary owner of DBus name - probably another PrusaSlicer instance is running.
[2020-12-20 17:44:18.468096] [0x00007ffff2124700] [error]   Dbus Messages listening terminating.
[Thread 0x7ffff2124700 (LWP 31780) exited]
[2020-12-20 17:44:18.561961] [0x00007ffff4e905c0] [info]    Detected gcode producer: Slic3r

Thread 1 "slic3r_main" hit Breakpoint 1, 0x0000000000ca7220 in Slic3r::GUI::GCodeViewer::log_memory_used(std::string const&, long) const ()
(gdb) c
Continuing.
[2020-12-20 17:44:23.082202] [0x00007ffff4e905c0] [trace]   Loaded G-code generated vertex buffers, 11MB Resident memory: 82MB; Shared memory: 41MB; Private memory: 41MB; Peak memory usage: 145MB

Thread 1 "slic3r_main" received signal SIGSEGV, Segmentation fault.
0x0000000000000000 in ?? ()
(gdb) 
```

Concatenating the gcode file four times makes it loadable (i suspect, that then the loading is slower and it behaves same as adding the breakpoint and continuing.

```
$ cat 3DBenchy_0.2mm_PETG_MK3S_1h27m.gcode 3DBenchy_0.2mm_PETG_MK3S_1h27m.gcode 3DBenchy_0.2mm_PETG_MK3S_1h27m.gcode 3DBenchy_0.2mm_PETG_MK3S_1h27m.gcode > 3d.gcode
$ PrusaSlicer-2.3.0-rc1+linux-x64-202012172035/bin/prusa-gcodeviewer 3d.gcode
# (Program works correctly)
```

lukasmatena, 2020-12-20:
I'm struggling to reproduce it. Could you please also provide a gcode that crashes for you? Thanks.

hadrava, 2020-12-20:
Here it is: I needed to add txt suffix for the github:
[3DBenchy_0.2mm_PETG_MK3S_1h27m.gcode.txt](https://github.com/prusa3d/PrusaSlicer/files/5721386/3DBenchy_0.2mm_PETG_MK3S_1h27m.gcode.txt)

sej7278, 2020-12-21:
looks like a duplicate of my #5464

hadrava, 2020-12-22:
I think this might help, everything was run on master from this afternoon (commit ea327c2d8cfd430dd301543400453003561af251)

First observation:
Segfault actually happens on line src/slic3r/GUI/GCodeViewer.cpp:1466 `glsafe(::glGenBuffers(1, &buffer.vertices.id));`

I am able to prevent it by adding breakpoint in gdb to basically any line in for-loop on lines 1397 - 1440 (or before the loop).

I was able to spot a difference in `m_buffers` after the for loop finishes (in both cases - with breakpoint in for-loop, so it was in OK state and one without the breakpoint so it was in NOK state).

In the OK state: the `m_buffers[0].shader = "options_120"` (and similarly for all other indices). But in NOK state all shader strings were empty. By further search, it looks like this values get assigned by  `Slic3r::GUI::GCodeViewer::render()` function.

So i tried to add breakpoint to this function and it seems to be correct way of looking. It is not called before the Segfault in the NOK variant.

But if I make the loop slower (larger gcode file, adding breakpoint in gdb), it gets called from the wxEvtHandler inside `progress_dialog->Update()`. Here is the stacktrace showing this:
```
(gdb) bt
#0  0x00005555560535c0 in Slic3r::GUI::GCodeViewer::render() const (this=this@entry=0x555557a40ec8) at /home/had/newprusa/PrusaSlicer-debinfo/src/slic3r/GUI/GCodeViewer.cpp:534
#1  0x0000555555fc61bc in Slic3r::GUI::GLCanvas3D::_render_gcode() const (this=this@entry=0x555557a40670) at /home/had/newprusa/PrusaSlicer-debinfo/src/slic3r/GUI/GLCanvas3D.cpp:5103
#2  0x0000555555fdd877 in Slic3r::GUI::GLCanvas3D::render() (this=0x555557a40670) at /home/had/newprusa/PrusaSlicer-debinfo/src/slic3r/GUI/GLCanvas3D.cpp:1659
#3  0x0000555556b6d3bf in wxAppConsoleBase::CallEventHandler(wxEvtHandler*, wxEventFunctor&, wxEvent&) const (this=0x5555574f3700, handler=0x555557b7a530, functor=..., event=...) at /home/had/newprusa/PrusaSlicer-debinfo/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/common/appbase.cpp:671
#4  0x0000555556c01cf8 in wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&) (entry=..., handler=0x555557b7a530, event=...) at /home/had/newprusa/PrusaSlicer-debinfo/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/common/event.cpp:1415
#5  0x0000555556c0295b in wxEvtHandler::SearchDynamicEventTable(wxEvent&) (this=0x555557b7a530, event=...) at /home/had/newprusa/PrusaSlicer-debinfo/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/common/event.cpp:1887
#6  0x0000555556c0211c in wxEvtHandler::TryHereOnly(wxEvent&) (this=0x555557b7a530, event=...) at /home/had/newprusa/PrusaSlicer-debinfo/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/common/event.cpp:1608
#7  0x0000555556c0348f in wxEvtHandler::TryBeforeAndHere(wxEvent&) (this=0x555557b7a530, event=...) at /home/had/newprusa/PrusaSlicer-debinfo/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/include/wx/event.h:3912
#8  0x0000555556c01f8f in wxEvtHandler::ProcessEventLocally(wxEvent&) (this=0x555557b7a530, event=...) at /home/had/newprusa/PrusaSlicer-debinfo/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/common/event.cpp:1545
#9  0x0000555556c01f26 in wxEvtHandler::ProcessEvent(wxEvent&) (this=0x555557b7a530, event=...) at /home/had/newprusa/PrusaSlicer-debinfo/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/common/event.cpp:1518
#10 0x0000555556c02217 in wxEvtHandler::SafelyProcessEvent(wxEvent&) (this=0x555557b7a530, event=...) at /home/had/newprusa/PrusaSlicer-debinfo/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/common/event.cpp:1636
#11 0x0000555556a23b40 in wxWindowBase::HandleWindowEvent(wxEvent&) const (this=0x555557b7a530, event=...) at /home/had/newprusa/PrusaSlicer-debinfo/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/common/wincmn.cpp:1555
#12 0x0000555556aaea40 in wxWindow::GTKSendPaintEvents(_GdkRegion const*) (this=0x555557b7a530, region=0x55555773f980) at /home/had/newprusa/PrusaSlicer-debinfo/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/gtk/window.cpp:5216
#13 0x0000555556aa54e7 in expose_event(GtkWidget*, GdkEventExpose*, wxWindow*) (gdk_event=0x7fffffff7460, win=0x555557b7a530) at /home/had/newprusa/PrusaSlicer-debinfo/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/gtk/window.cpp:432
#14 0x00007ffff6c5c1eb in  () at /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0
#15 0x00007ffff73f2c8d in g_closure_invoke () at /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#16 0x00007ffff7406365 in  () at /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#17 0x00007ffff740e9ab in g_signal_emit_valist () at /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#18 0x00007ffff740f97f in g_signal_emit () at /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#19 0x00007ffff6d72cac in  () at /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0
#20 0x00007ffff6c5aa81 in gtk_main_do_event () at /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0
#21 0x0000555556a95e63 in wxgtk_main_do_event(GdkEvent*, void*) (event=0x7fffffff7460, data=0x555557acfba0) at /home/had/newprusa/PrusaSlicer-debinfo/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/gtk/evtloop.cpp:359
#22 0x00007ffff6fb280f in  () at /usr/lib/x86_64-linux-gnu/libgdk-x11-2.0.so.0
#23 0x00007ffff6faf0cd in  () at /usr/lib/x86_64-linux-gnu/libgdk-x11-2.0.so.0
#24 0x00007ffff6faf9e3 in gdk_window_process_all_updates () at /usr/lib/x86_64-linux-gnu/libgdk-x11-2.0.so.0
#25 0x00007ffff6be0c11 in  () at /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0
[... cut ...]
By further investigation, i found that if it runs correctly: all following events gets called before the segfault place (yes, in this order)
```
wxTimerEvent
wxCloseEvent
wxPaintEvent
```
But they are not called before segfault. (This list might not be full, there may also be some other wx events in which are the two runs different.)

On the other hand, if i comment out progress_dialog update on lines 1403 - 1408 in `GCodeViewer.cpp`. None of the events gets handled and program segfaults even on large .gcodes or manual breakpoints. So this modification might be a way for you to reproduce issue on your machine.

hadrava, 2020-12-22:
Perfect, I am not able to reproduce the issue with commit 81878a6.

lukasmatena, 2020-12-22:
@hadrava Thanks, I'll close the issue then. And thanks again for your help during the debugging.

closed 2020-12-22 by lukasmatena