prusa3d/PrusaSlicer #5538: User gcode breaks the graphics of sliced model

KRUP85, 2020-12-20:
### Version
2.3.0 rc1, Windows 10 home, 20H2

When I added these two commands to gcode "before layer change", the matched model is displayed incorrectly. However, gcode is fine.

;BEFORE_LAYER_CHANGE
G92 E0.0
;LAYER:[layer_num]
;HEIGHT:[layer_z]

[image]
[image]
[image]
[image]

rtyr, 2020-12-20:
Similar to https://github.com/prusa3d/PrusaSlicer/issues/4992.

bubnikv, 2021-01-05:
We need to check the custom G-code sections for the annotations that PrusaSlicer utilizes for its G-code viewer and refuse to slice with such custom G-code sections, or at least give a warning.

;LAYER:xxx
;HEIGHT:xxx

bubnikv, 2021-08-06:
PrusaSlicer 2.4.0-alpha1 will test for the reserved comments in all custom G-codes and such custom G-codes will not be accepted.

closed 2021-08-06 by bubnikv