prusa3d/PrusaSlicer #5541: Scale and alignment of attached parts lost on 3mf reload

bartrv, 2020-12-20:
### Version
2.3.0-rc1

### Operating system type + version
Windows 10 x64

### 3D printer brand / version 
MK3S+MMU2s

### Behavior
When using "Add Part...-> Load" on an object scaled in slicer the default behavior is to translate the imported stl part to the position, scale and orientation of the parent object it is added to. Which is great.  If there is an update to an attached part, the stl can be overwritten and "Reload(ed) from disk", the update swaps in seamlessly.

- the Problem:
If the 3mf file is closed and reopened (both by closing/re-launching slic3r AND by opening a different 3mf file and then re-opening the original file) the reload from disk looses part of its alignment.  The x, y , scale, and orientation are maintained, but the 'z' alignment is lost and the reloaded stl appears floating above the object it is attached to.

_Is this a new feature request?_
Nope

lukasmatena, 2020-12-20:
Reproduced, thanks. The issue is already present in 2.2.0 release.

FidelCapo, 2021-01-28:
This issue should be fixed in next release.
Closing.

closed 2021-01-28 by FidelCapo