prusa3d/PrusaSlicer #5547: XY size compensation for 0 value.

Shama71, 2020-12-21:
### Version
2.3.0-rc1

### Operating system type + version
Windows 10 x64
### Behavior
XY size compensation for value 0, after slicing appears to be negative.

[image]
[image]

neophyl, 2020-12-21:
That looks like the first layer, which will be using an elephants foot value of 0.2 by default. That moves the first layer in by the specified amount to offset and first layer expansion caused by ‘squish’ to get a good first layer adhesion.

Shama71, 2020-12-21:
Yes, I really have a value of 0.2 for Elephant foot compensationю

rtyr, 2020-12-21:
OK. Could you please close this issue?

closed 2020-12-21 by Shama71