prusa3d/PrusaSlicer #5548: Concentric infill needs to fill the center point

Zunzulla, 2020-12-21:
### Version
PrusaSlicer 2.3.0-rc1+win64

### Behavior
When using concentric top fill / bottom fill pattern there is always a gap in the center of the circle. This could be improved by extruding some plastic into the center point. 

This can be seen as a feature request (new pattern option) or improvement suggestion. This is probably not a bug

[image]

[Shape-Cylinder.zip](https://github.com/prusa3d/PrusaSlicer/files/5724307/Shape-Cylinder.zip)

rtyr, 2020-12-21:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/577. Closing.

closed 2020-12-21 by rtyr

Zunzulla, 2020-12-21:
@rtyr I've already looked for duplicates. #577 represents circular extrusions missing in the infill pattern. I am talking about the center not being filled due to not enough space for fitting an extrusion. The feature to fill the center exists in superslicer (concentric filled) but not in prusaslicer

rtyr, 2020-12-21:
It is still the same issue.

Zunzulla, 2020-12-21:
@rtyr alright thanks a lot! Just out of curiosity regarding the dates: this will not be worked on as far as I can tell?

rtyr, 2020-12-21:
I believe it will eventually.