prusa3d/PrusaSlicer #5549: Slicer doesn't see all windows drives when trying to add STL

Jester7896, 2020-12-21:
### Version
_Version 2.3.0-rc1+win64_

### Operating system type + version
Windows 10 Home 20H2

### 3D printer brand / version + firmware version (if known)
MK3S

### Behavior
I have a networked drive mapped to Z:\ where I store my STL's 

When I try to add an STL or open a project the slicer does not list that drive to navigate to.

 Windows Explorer

[image]

Slicer Open Menu

[image]

neophyl, 2020-12-21:
Just checked with the rc1 at work and I can access all the mapped and non mapped drives on the network.  We are running a different version of Windows though -  Win 10 Pro Build 18363.
This would suggest that its a windows problem and not Prusa Slicer.

Jester7896, 2020-12-21:
So I should have done my due diligence with a basic google search on this phenomenon.  I got in such a habit to always "Run as administrator" since nearly all my work software requires this.  If I do not choose to run the program as an administrator it works as I would expect.

closed 2020-12-21 by Jester7896