prusa3d/PrusaSlicer #5550: 2.3 rc1 can't load its own 3mf files (if they contain custom gcode)

sej7278, 2020-12-21:
adding custom layer change gcode (even a single line comment!) means that prusaslicer segfaults when trying to load it again.

i can import it as a project, but can't open it as a project. gcode-viewer can load the generated gcode. gtk2 and gtk3 make no difference. debian sid.

strace doesn't reveal much:

```
--- SIGSEGV {si_signo=SIGSEGV, si_code=SEGV_MAPERR, si_addr=NULL} ---
+++ killed by SIGSEGV +++
Segmentation fault
```

gdb isn't much better:

```
Thread 1 "slic3r_main" received signal SIGSEGV, Segmentation fault.
0x0000000000491773 in __gnu_cxx::__enable_if<std::__is_char<char>::__value, bool>::__type std::operator==<char>(std::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) ()
```

stl, 3mf, custom gcode and generated gcode attached: [test.zip](https://github.com/prusa3d/PrusaSlicer/files/5724956/test.zip)

lukasmatena, 2020-12-21:
I cannot reproduce it. Are we talking about our official build of [2.3.0-rc1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.3.0-rc1)? What exactly should I do to provoke the error? Does it crash immediately after you try to open the 3MF that you provided? Thanks.

sej7278, 2020-12-21:
yes tried both of these official builds:

PrusaSlicer-2.3.0-rc1+linux-x64-GTK3-202012172045.tar.bz2
PrusaSlicer-2.3.0-rc1+linux-x64-202012172035.tar.bz2

to reproduce simply open the file from file->open project, or run prusa-slicer test.3mf

it opens the window, you get the "loading..." popup then it closes the window

this is proper debian not wsl2 or anything

lukasmatena, 2020-12-21:
Thanks. Still can't reproduce though. If there is nothing sensitive in your ~/.config/PrusaSlicer directory, could you pack it and send it as well?

sej7278, 2020-12-21:
there's something odd with my printer profile i suspect as if i load the same 3mf into a debian buster vm it opens it fine until i import my printer config bundle. [~/.config/PrusaSlicer](https://github.com/prusa3d/PrusaSlicer/files/5725480/config.zip)

sej7278, 2020-12-22:
just fyi the PrusaSlicer-2.3.0-rc1+28-linux-x64-GTK3-g81878a6-202012221403 build didn't fix this (not that it was supposed to)

sej7278, 2020-12-22:
setting a bunch of breakpoints one at a time in gdb, i seem to be able to get as far as `Slic3r::GUI::GUI_App::load_current_presets()` before it segfaults:

```
Thread 1 "slic3r_main" hit Breakpoint 1, 0x0000000000983ec0 in Slic3r::GUI::GUI_App::load_current_presets(bool) ()
(gdb) c
Continuing.
......

Thread 1 "slic3r_main" received signal SIGSEGV, Segmentation fault.
0x0000000000491773 in __gnu_cxx::__enable_if<std::__is_char<char>::__value, bool>::__type std::operator==<char>(std::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) ()
(gdb) bt
#0  0x0000000000491773 in __gnu_cxx::__enable_if<std::__is_char<char>::__value, bool>::__type std::operator==<char>(std::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) ()
#1  0x0000000000d2b329 in void Slic3r::add_correct_opts_to_diff<Slic3r::ConfigOptionStrings>(std::string const&, std::vector<std::string, std::allocator<std::string> >&, Slic3r::ConfigBase const&, Slic3r::ConfigBase const&) ()
#2  0x0000000000d2bd50 in Slic3r::deep_diff(Slic3r::ConfigBase const&, Slic3r::ConfigBase const&) ()
#3  0x0000000000d161d5 in Slic3r::PresetCollection::dirty_options(Slic3r::Preset const*, Slic3r::Preset const*, bool) ()
#4  0x0000000000a67217 in Slic3r::GUI::Tab::update_changed_ui() ()
#5  0x0000000000a67749 in Slic3r::GUI::Tab::load_current_preset() ()
#6  0x0000000000983f40 in Slic3r::GUI::GUI_App::load_current_presets(bool) ()
#7  0x00000000009e9ee6 in Slic3r::GUI::Plater::priv::load_files(std::vector<boost::filesystem::path, std::allocator<boost::filesystem::path> > const&, bool, bool, bool) ()
#8  0x00000000009ec00b in Slic3r::GUI::Plater::load_files(std::vector<boost::filesystem::path, std::allocator<boost::filesystem::path> > const&, bool, bool, bool) ()
#9  0x00000000009ec182 in Slic3r::GUI::Plater::load_project(wxString const&) ()
#10 0x00000000009ec30e in Slic3r::GUI::Plater::load_project() ()
#11 0x0000000001832827 in wxAppConsoleBase::CallEventHandler(wxEvtHandler*, wxEventFunctor&, wxEvent&) const (this=0x23815e0, handler=0x300c870, functor=warning: RTTI symbol not found for class 'wxEventFunctorFunctor<wxEventTypeTag<wxCommandEvent>, std::function<void (wxCommandEvent&)> >'
..., event=...)
    at /home/buildbot/PrusaSlicer/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/common/appbase.cpp:671
#12 0x00000000018c6eda in wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&) (entry=..., handler=0x300c870, event=...)
    at /home/buildbot/PrusaSlicer/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/common/event.cpp:1415
#13 0x00000000018c7b3b in wxEvtHandler::SearchDynamicEventTable(wxEvent&) (this=0x300c870, event=...) at /home/buildbot/PrusaSlicer/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/common/event.cpp:1887
#14 0x00000000018c72fe in wxEvtHandler::TryHereOnly(wxEvent&) (this=0x300c870, event=...) at /home/buildbot/PrusaSlicer/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/common/event.cpp:1608
#15 0x00000000018c868f in wxEvtHandler::TryBeforeAndHere(wxEvent&) (this=0x300c870, event=...) at /home/buildbot/PrusaSlicer/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/include/wx/event.h:3912
#16 0x00000000018c716f in wxEvtHandler::ProcessEventLocally(wxEvent&) (this=0x300c870, event=...) at /home/buildbot/PrusaSlicer/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/common/event.cpp:1545
#17 0x00000000018c7106 in wxEvtHandler::ProcessEvent(wxEvent&) (this=0x300c870, event=...) at /home/buildbot/PrusaSlicer/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/common/event.cpp:1518
#18 0x00000000018c73f9 in wxEvtHandler::SafelyProcessEvent(wxEvent&) (this=0x300c870, event=...) at /home/buildbot/PrusaSlicer/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/common/event.cpp:1636
#19 0x00000000016a5a88 in wxMenuBase::DoProcessEvent(wxMenuBase*, wxEvent&, wxWindow*) (menu=0x300c870, event=..., win=0x2816c30) at /home/buildbot/PrusaSlicer/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/common/menucmn.cpp:667
#20 0x00000000016a59bb in wxMenuBase::SendEvent(int, int) (this=0x300c870, itemid=179, checked=-1) at /home/buildbot/PrusaSlicer/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/common/menucmn.cpp:643
#21 0x00000000017b7f82 in menuitem_activate(GtkWidget*, wxMenuItem*) (item=0x3012cf0) at /home/buildbot/PrusaSlicer/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/gtk/menu.cpp:541
#22 0x00007ffff6e84092 in g_closure_invoke () at /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#23 0x00007ffff6e96403 in  () at /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#24 0x00007ffff6e9c69f in g_signal_emit_valist () at /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#25 0x00007ffff6e9cc0f in g_signal_emit () at /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#26 0x00007ffff7728424 in gtk_widget_activate () at /usr/lib/x86_64-linux-gnu/libgtk-3.so.0
#27 0x00007ffff75ee5a6 in gtk_menu_shell_activate_item () at /usr/lib/x86_64-linux-gnu/libgtk-3.so.0
#28 0x00007ffff75ee883 in  () at /usr/lib/x86_64-linux-gnu/libgtk-3.so.0
#29 0x00007ffff777bac8 in  () at /usr/lib/x86_64-linux-gnu/libgtk-3.so.0
#30 0x00007ffff6e842de in  () at /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#31 0x00007ffff6e9bdc9 in g_signal_emit_valist () at /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
[... cut ...]
rip            0x491773            0x491773 <__gnu_cxx::__enable_if<std::__is_char<char>::__value, bool>::__type std::operator==<char>(std::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)+3>
eflags         0x10202             [ IF RF ]
cs             0x33                51
ss             0x2b                43
ds             0x0                 0
es             0x0                 0
fs             0x0                 0
gs             0x0                 0
```

lukasmatena, 2020-12-23:
@sej7278 Thanks, this is helpful. I still cannot reproduce it, but I don't like it.
1. When did this problem start to appear? Was it ok in 2.3.0-beta3?
2. Can you add following debug output and post the end of it before the crash?

```diff
--- a/src/libslic3r/Preset.cpp
+++ b/src/libslic3r/Preset.cpp
@@ -1134,6 +1134,21 @@ void add_correct_opts_to_diff(const std::string &opt_key, t_config_option_keys&
     const T* opt_init = static_cast<const T*>(other.option(opt_key));
     const T* opt_cur = static_cast<const T*>(this_c.option(opt_key));
     int opt_init_max_id = opt_init->values.size() - 1;
+
+    if (std::is_same<T, ConfigOptionStrings>::value) {
+        auto cur = dynamic_cast<const ConfigOptionStrings*>(opt_cur);
+        auto init = dynamic_cast<const ConfigOptionStrings*>(opt_init);
+
+        std::cout << "-----------\n" << "opt_key: " << opt_key << std::endl;
+        std::cout << "opt_cur(" << cur->values.size() << "):" << std::endl;
+        for (const std::string& s : cur->values)
+            std::cout << "    " << s << std::endl;
+        std::cout << "opt_init(" << init->values.size() << "):" << std::endl;
+        for (const std::string& s : init->values)
+            std::cout << "    " << s << std::endl;
+        std::cout << "-----------" << std::endl;
+    }
+
     for (int i = 0; i < int(opt_cur->values.size()); i++)
     {
         int init_id = i <= opt_init_max_id ? i : 0;

```

sej7278, 2020-12-23:
beta1 doesn't have the problem, beta2 and beta3 do (i copied ~/.config/PrusaSlicer to PrusaSlicer-beta)

did you try with my settings above - it only seems to happen once i've imported those - looks like something to do with changing some config value from the default preset?

i'll look to see how to compile it in the morning to apply the diff.

lukasmatena, 2020-12-23:
> i'll look to see how to compile it in the morning to apply the diff.

Sorry, I somehow thought that you already had the build set up, I didn't mean to make you. I can later post a build for you to test later. And yes, I'm using your config and loading your test.3mf. I'll go through the changes after beta1. I'd like to know what's going on after seeing the backtrace.

sej7278, 2020-12-23:
i did try to compile it but there seems like an endless dependency chain on debian - libboost-all-dev, libtbb-dev, openssl and then all the wx stuff.... and i didn't really know which options (static etc.) so gave up i'm afraid

sej7278, 2020-12-23:
@lukasmatena  i managed to compile it in debug mode with your patch, hopefully this means more to you than me:

```
(prusa-slicer:2155735): Gtk-CRITICAL **: 16:59:09.484: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkScrollbar
[debug] semver.c 2.2.0 = 2.2.0, (null) (null)
[Thread 0x7fffc37fe700 (LWP 2155753) exited]
[debug] semver.c 2.2.0 = 2.2.0, (null) (null)
[debug] semver.c 2.3.0-rc2 = 2.3.0, rc2 (null)
[New Thread 0x7fffc2ffd700 (LWP 2155762)]
[New Thread 0x7fffc1dfc700 (LWP 2155763)]
[New Thread 0x7fffc15fb700 (LWP 2155764)]
[New Thread 0x7fffc0dfa700 (LWP 2155765)]
[New Thread 0x7fffa7fff700 (LWP 2155766)]
[Thread 0x7fffc15fb700 (LWP 2155764) exited]
[Thread 0x7fffc1dfc700 (LWP 2155763) exited]
[Thread 0x7fffc2ffd700 (LWP 2155762) exited]
[Thread 0x7fffc3fff700 (LWP 2155752) exited]
[New Thread 0x7fffc3fff700 (LWP 2155775)]
[Thread 0x7fffc3fff700 (LWP 2155775) exited]
[New Thread 0x7fffc3fff700 (LWP 2155776)]
[Thread 0x7fffc0dfa700 (LWP 2155765) exited]
-----------
opt_key: default_filament_profile
opt_cur(1):
    
opt_init(0):
-----------

Thread 1 "slic3r_main" received signal SIGSEGV, Segmentation fault.
0x00007ffff7ee4c50 in std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::size() const () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
(gdb) bt
#0  0x00007ffff7ee4c50 in std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::size() const () at /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#1  0x00005555556ce755 in std::operator==<char>(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) (__lhs="", __rhs=<error reading variable: Cannot access memory at address 0x8>) at /usr/include/c++/10/bits/basic_string.h:6173
#2  0x00005555556d80a0 in std::operator!=<char, std::char_traits<char>, std::allocator<char> >(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) (__lhs="", __rhs=<error reading variable: Cannot access memory at address 0x8>)
    at /usr/include/c++/10/bits/basic_string.h:6242
#3  0x00005555566e364b in Slic3r::add_correct_opts_to_diff<Slic3r::ConfigOptionStrings>(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >&, Slic3r::ConfigBase const&, Slic3r::ConfigBase const&) (opt_key="default_filament_profile", vec=std::vector of length 1, capacity 1 = {...}, other=..., this_c=...)
    at /data3/useful_software/3dprinting/slic3r/git/src/libslic3r/Preset.cpp:1155
#4  0x00005555566e1b08 in Slic3r::deep_diff[abi:cxx11](Slic3r::ConfigBase const&, Slic3r::ConfigBase const&) (config_this=..., config_other=...)
    at /data3/useful_software/3dprinting/slic3r/git/src/libslic3r/Preset.cpp:1178
#5  0x00005555566d82cf in Slic3r::PresetCollection::dirty_options[abi:cxx11](Slic3r::Preset const*, Slic3r::Preset const*, bool)
    (edited=0x555558001b58, reference=0x5555582872a0, deep_compare=true) at /data3/useful_software/3dprinting/slic3r/git/src/libslic3r/Preset.cpp:1192
#6  0x0000555556211533 in Slic3r::PresetCollection::current_dirty_options[abi:cxx11](bool) const (this=0x555558001ab0, deep_compare=true)
    at /data3/useful_software/3dprinting/slic3r/git/src/libslic3r/Preset.hpp:438
#7  0x00005555562879bd in Slic3r::GUI::Tab::update_changed_ui() (this=0x555558bff1b0) at /data3/useful_software/3dprinting/slic3r/git/src/slic3r/GUI/Tab.cpp:605
#8  0x00005555562ab88a in Slic3r::GUI::Tab::load_current_preset() (this=0x555558bff1b0) at /data3/useful_software/3dprinting/slic3r/git/src/slic3r/GUI/Tab.cpp:2937
#9  0x0000555556118cb3 in Slic3r::GUI::GUI_App::load_current_presets(bool) (this=0x555557e83f90, check_printer_presets_=false)
    at /data3/useful_software/3dprinting/slic3r/git/src/slic3r/GUI/GUI_App.cpp:1852
#10 0x00005555561a19fd in Slic3r::GUI::Plater::priv::load_files(std::vector<boost::filesystem::path, std::allocator<boost::filesystem::path> > const&, bool, bool, bool) (this=
    0x55555843e400, input_files=std::vector of length 1, capacity 1 = {...}, load_model=true, load_config=true, imperial_units=false)
    at /data3/useful_software/3dprinting/slic3r/git/src/slic3r/GUI/Plater.cpp:2382
#11 0x00005555561b433d in Slic3r::GUI::Plater::load_files(std::vector<boost::filesystem::path, std::allocator<boost::filesystem::path> > const&, bool, bool, bool)
    (this=0x555557ebccd0, input_files=std::vector of length 1, capacity 1 = {...}, load_model=true, load_config=true, imperial_units=false)
    at /data3/useful_software/3dprinting/slic3r/git/src/slic3r/GUI/Plater.cpp:4888
#12 0x00005555561b30cd in Slic3r::GUI::Plater::load_project(wxString const&) (this=0x555557ebccd0, filename=...) at /data3/useful_software/3dprinting/slic3r/git/src/slic3r/GUI/Plater.cpp:4775
#13 0x00005555561b2ea7 in Slic3r::GUI::Plater::load_project() (this=0x555557ebccd0) at /data3/useful_software/3dprinting/slic3r/git/src/slic3r/GUI/Plater.cpp:4759
#14 0x000055555615d585 in operator()(wxCommandEvent&) const (__closure=0x555558c8b3c8) at /data3/useful_software/3dprinting/slic3r/git/src/slic3r/GUI/MainFrame.cpp:940
[... cut ...]
#52 0x0000555556108736 in Slic3r::GUI::GUI_Run(Slic3r::GUI::GUI_InitParams&) (params=...) at /data3/useful_software/3dprinting/slic3r/git/src/slic3r/GUI/GUI_Init.cpp:69
#53 0x00005555556a5f83 in Slic3r::CLI::run(int, char**) (this=0x7fffffffdf20, argc=1, argv=0x7fffffffe138) at /data3/useful_software/3dprinting/slic3r/git/src/PrusaSlicer.cpp:576
#54 0x00005555556a7d7f in main(int, char**) (argc=1, argv=0x7fffffffe138) at /data3/useful_software/3dprinting/slic3r/git/src/PrusaSlicer.cpp:777
(gdb) 
```

sej7278, 2020-12-23:
added some debugging of my own to list the options being processed.

this is a flashforge creator pro but only the left extruder (2) is being used by this petg profile, only abs profiles use the right extruder (1). not sure if 2 extruders is relevant but thought it might help.

```diff
@@ -1147,6 +1162,8 @@ inline t_config_option_keys deep_diff(const ConfigBase &config_this, const Confi
 {
     t_config_option_keys diff;
     for (const t_config_option_key &opt_key : config_this.keys()) {
+        std::cout << "2-----------\n" << "opt_key: " << opt_key << std::endl;
+
         const ConfigOption *this_opt  = config_this.option(opt_key);
         const ConfigOption *other_opt = config_other.option(opt_key);
         if (this_opt != nullptr && other_opt != nullptr && *this_opt != *other_opt)
```

```
2-----------
opt_key: bed_custom_model
2-----------
opt_key: bed_custom_texture
2-----------
opt_key: bed_shape
2-----------
opt_key: before_layer_gcode
2-----------
opt_key: between_objects_gcode
2-----------
opt_key: color_change_gcode
2-----------
opt_key: cooling_tube_length
2-----------
opt_key: cooling_tube_retraction
2-----------
opt_key: default_filament_profile
-----------
opt_key: default_filament_profile
opt_cur(1):
    
opt_init(0):
-----------
```

lukasmatena, 2020-12-23:
There is something suspicious. Could you please try the following? Thanks.

```diff
--- a/src/libslic3r/Preset.cpp
+++ b/src/libslic3r/Preset.cpp
@@ -283,7 +283,7 @@ void Preset::normalize(DynamicPrintConfig &config)
                 static_cast<ConfigOptionVectorBase*>(opt)->resize(n, defaults.option(key));
         }
         // The following keys are mandatory for the UI, but they are not part of FullPrintConfig, therefore they are handled separately.
-        for (const std::string &key : { "filament_settings_id" }) {
+        for (const std::string &key : { "filament_settings_id", "default_filament_profile" }) {
             auto *opt = config.option(key, false);
             assert(opt == nullptr || opt->type() == coStrings);
             if (opt != nullptr && opt->type() == coStrings)

```

sej7278, 2020-12-23:
here you go:

```
[2020-12-23 23:09:50.657717] [0x00007ffff0fb2e40] [trace]   TriangleMeshSlicer::require_shared_vertices - end
2-----------
opt_key: bed_custom_model
2-----------
opt_key: bed_custom_texture
2-----------
opt_key: bed_shape
2-----------
opt_key: before_layer_gcode
2-----------
opt_key: between_objects_gcode
2-----------
opt_key: color_change_gcode
2-----------
opt_key: cooling_tube_length
2-----------
opt_key: cooling_tube_retraction
2-----------
opt_key: default_filament_profile
-----------
opt_key: default_filament_profile
opt_cur(1):
    
opt_init(0):
-----------

Thread 1 "slic3r_main" received signal SIGSEGV, Segmentation fault.
0x00007ffff7ee4c50 in std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::size() const () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
(gdb) 
(gdb) 
(gdb) 
(gdb) 
(gdb) bt
#0  0x00007ffff7ee4c50 in std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::size() const () at /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#1  0x00005555556ce755 in std::operator==<char>(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
    (__lhs="", __rhs=<error reading variable: Cannot access memory at address 0x8>) at /usr/include/c++/10/bits/basic_string.h:6173
#2  0x00005555556d80a0 in std::operator!=<char, std::char_traits<char>, std::allocator<char> >(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) (__lhs="", __rhs=<error reading variable: Cannot access memory at address 0x8>) at /usr/include/c++/10/bits/basic_string.h:6242
#3  0x00005555566e3689 in Slic3r::add_correct_opts_to_diff<Slic3r::ConfigOptionStrings>(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >&, Slic3r::ConfigBase const&, Slic3r::ConfigBase const&)
    (opt_key="default_filament_profile", vec=std::vector of length 1, capacity 1 = {...}, other=..., this_c=...) at /data3/useful_software/3dprinting/slic3r/git/src/libslic3r/Preset.cpp:1155
#4  0x00005555566e1b46 in Slic3r::deep_diff[abi:cxx11](Slic3r::ConfigBase const&, Slic3r::ConfigBase const&) (config_this=..., config_other=...) at /data3/useful_software/3dprinting/slic3r/git/src/libslic3r/Preset.cpp:1180
#5  0x00005555566d82c1 in Slic3r::PresetCollection::dirty_options[abi:cxx11](Slic3r::Preset const*, Slic3r::Preset const*, bool) (edited=0x555557f56198, reference=0x55555829c5c0, deep_compare=true)
    at /data3/useful_software/3dprinting/slic3r/git/src/libslic3r/Preset.cpp:1194
#6  0x0000555556211533 in Slic3r::PresetCollection::current_dirty_options[abi:cxx11](bool) const (this=0x555557f560f0, deep_compare=true) at /data3/useful_software/3dprinting/slic3r/git/src/libslic3r/Preset.hpp:438
#7  0x00005555562879bd in Slic3r::GUI::Tab::update_changed_ui() (this=0x555558c00ba0) at /data3/useful_software/3dprinting/slic3r/git/src/slic3r/GUI/Tab.cpp:605
#8  0x00005555562ab88a in Slic3r::GUI::Tab::load_current_preset() (this=0x555558c00ba0) at /data3/useful_software/3dprinting/slic3r/git/src/slic3r/GUI/Tab.cpp:2937
#9  0x0000555556118cb3 in Slic3r::GUI::GUI_App::load_current_presets(bool) (this=0x555557e85e40, check_printer_presets_=false) at /data3/useful_software/3dprinting/slic3r/git/src/slic3r/GUI/GUI_App.cpp:1852
#10 0x00005555561a19fd in Slic3r::GUI::Plater::priv::load_files(std::vector<boost::filesystem::path, std::allocator<boost::filesystem::path> > const&, bool, bool, bool) (this=
    0x555558440400, input_files=std::vector of length 1, capacity 1 = {...}, load_model=true, load_config=true, imperial_units=false) at /data3/useful_software/3dprinting/slic3r/git/src/slic3r/GUI/Plater.cpp:2382
#11 0x00005555561b433d in Slic3r::GUI::Plater::load_files(std::vector<boost::filesystem::path, std::allocator<boost::filesystem::path> > const&, bool, bool, bool)
    (this=0x555557fca0d0, input_files=std::vector of length 1, capacity 1 = {...}, load_model=true, load_config=true, imperial_units=false) at /data3/useful_software/3dprinting/slic3r/git/src/slic3r/GUI/Plater.cpp:4888
#12 0x00005555561b30cd in Slic3r::GUI::Plater::load_project(wxString const&) (this=0x555557fca0d0, filename=...) at /data3/useful_software/3dprinting/slic3r/git/src/slic3r/GUI/Plater.cpp:4775
#13 0x00005555561b2ea7 in Slic3r::GUI::Plater::load_project() (this=0x555557fca0d0) at /data3/useful_software/3dprinting/slic3r/git/src/slic3r/GUI/Plater.cpp:4759
#14 0x000055555615d585 in operator()(wxCommandEvent&) const (__closure=0x555558c8b3f8) at /data3/useful_software/3dprinting/slic3r/git/src/slic3r/GUI/MainFrame.cpp:940
#15 0x0000555556183c1d in std::__invoke_impl<void, Slic3r::GUI::MainFrame::init_menubar_as_editor()::<lambda(wxCommandEvent&)>&, wxCommandEvent&>(std::__invoke_other, struct {...} &) (__f=...) at /usr/include/c++/10/bits/invoke.h:60
#16 0x000055555617a23a in std::__invoke_r<void, Slic3r::GUI::MainFrame::init_menubar_as_editor()::<lambda(wxCommandEvent&)>&, wxCommandEvent&>(struct {...} &) (__fn=...) at /usr/include/c++/10/bits/invoke.h:110
[... cut ...]
#51 0x00007ffff6f4ca19 in wxEntry(int&, wchar_t**) () at /usr/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#52 0x0000555556108736 in Slic3r::GUI::GUI_Run(Slic3r::GUI::GUI_InitParams&) (params=...) at /data3/useful_software/3dprinting/slic3r/git/src/slic3r/GUI/GUI_Init.cpp:69
#53 0x00005555556a5f83 in Slic3r::CLI::run(int, char**) (this=0x7fffffffdf00, argc=3, argv=0x7fffffffe118) at /data3/useful_software/3dprinting/slic3r/git/src/PrusaSlicer.cpp:576
#54 0x00005555556a7d7f in main(int, char**) (argc=3, argv=0x7fffffffe118) at /data3/useful_software/3dprinting/slic3r/git/src/PrusaSlicer.cpp:777
```

lukasmatena, 2020-12-23:
Ok, thank you very much. I see where it crashes but I don't understand how it got there and why is it so difficult to reproduce. We'll investigate the issue further when there is more time.

sej7278, 2020-12-23:
If i don't import my profile and just use the default prusa mini, i can't reproduce this.

if i import my profile and don't add any custom gcode its fine.

so something to do with a non-default profile (that doesn't have custom gcode to begin with) and adding custom gcode it seems.

anyway i guess its low priority in the grand scheme of things, thanks for looking into it and happy christmas!

sej7278, 2020-12-28:
@lukasmatena looking a bit more into it, it seems `default_filament_profile` should be set to an empty string (not null) in the Slic3r_PE.config file inside the 3mf archive.

at the moment its set to:
```
; default_filament_profile = 
```

if i change that to:
```
; default_filament_profile = ""
```
it doesn't crash, but it doesn't select the correct user printer profile - should be "FFCP single material L" in my example.

DrLex0, 2020-12-28:
I have never done anything with that parameter in the profiles and I don't even know what it's for. It's just there, and in almost all the gcode files that I can find, it has an empty value:
```
; default_filament_profile = 
```
Only in a few old files created with 2.0.0-rc, I find an empty string.
I don't encounter any crashes though in the Mac OS version when opening 3mf files that use that exact same profile…

sej7278, 2020-12-29:
@DrLex0 yes, so looks like its a parser issue - doesn't show up on built-in profiles that i guess most people use, just custom ones (yours is the only one i tried) did you try 2.3.0-rc2 on macos? i can't blame gtk3 this time as the gtk2 builds have the same issue on debian buster/bullseye.

p.s. i meant no offence - your gcode snippets are the only reason i can use slic3r instead of flashprint and i've printed pretty much every creator pro addon you designed ;-)

DrLex0, 2020-12-29:
@sej7278 No offence taken :) I just wanted to add some information. And yes, this was in RC2. I have opened 3MF files saved in both the previous betas and RC1, as well as RC2, and so far I haven't encountered any crashes…

sp847242, 2020-12-31:
I'm having this too with 2.3.0 RC2, Win10 Pro. Some 3MF files saved by PrusaSlicer cause it to either hang when loading them, or else cause it to close before any geometry loads, and with no error message. No processor usage either when it's frozen, and the memory usage does not change.
One such file is attached.

[tailstock_r4_top.zip](https://github.com/prusa3d/PrusaSlicer/files/5758263/tailstock_r4_top.zip)

sej7278, 2020-12-31:
@sp847242 thank goodness for that, thought I was losing my mind! Can you reproduce with my test.zip above? That's basically a tiny cube

sp847242, 2020-12-31:
@sej7278
Bad(?) news: Your file loaded normally for me. :) I tried all these methods:
- Double-click the .3MF file to open PrusaSlicer.
- Drag into PS (and tried each of the three Actions there).
- Open PS, then File -->Open Project.
The filament/printer settings all look to have updated accordingly, and there's a lot of custom G-code.

I also did File --> Import --> Import Config from Project. That worked normally too.
But if I try that Import Config with my .3MF, PS briefly tries processing it and then crashes itself closed.

sej7278, 2020-12-31:
Weird, so it's not particularly reproducible, I'll try your files in a bit.

Only thing that definitely fixes it for me is setting:

; default_filament_profile = ""

either in my printer settings or in the 3mf if it contains custom gcode, there's definitely something wrong with config parsing after beta1

sp847242, 2020-12-31:
I just tried your method on my .3MF file: Changed _;default_filament_profile =_ to _;default_filament_profile = ""_ and then re-zipped the files and changed it back to a .3MF extension.
Same results as before.

And it does appear consistent in this regard:

- File --> Open Project results in crash/close-program.
- Dragging the .3MF file into PS results in a program lockup with the "Loading..." box still visible.
- Double-clicking a .3MF file to open PS results in crash/close-program.

sej7278, 2020-12-31:
equally as weird - i can't get your 2 files to crash at all.

sp847242, 2020-12-31:
Huh. Well thank you for trying them anyway.

Maybe these are actually two separate issues then. Either way, at least for me, .3MF files are mostly unusable in 2.3.0 rc2+win64.
...
I just tried 5 randomly-selected .3MF files made in the past year. Only one of them opened normally.

v2.2.0+win64 opens them all fine. (Including both .3MFs with and without the default_filament_profile modification.)

bubnikv, 2021-01-04:
I am seeing an out of bounds access when trying to load the very first 3mf provided by @sej7278 over configs provided by @sej7278. That is a good news.

bubnikv, 2021-01-04:
@sp847242 Could you please share the content of your PrusaSlicer configuration directory?

bubnikv, 2021-01-04:
What changed is the way the profile inheritance is handled after loading a 3MF file. Before the profile from a 3MF file has been loaded as its own profile, showing the "modified" status against itself, thus not modified initially. Now the profile from the 3MF is being loaded over a local profile, from which it has been derived, thus some of the fields may initially be marked as "modified" against the parent profile. 

The bug now is in handling some of the vector variables (dependent on number of extruders), which may not contain the correct number of elements initially.

bubnikv, 2021-01-04:
Hopefully fixed with https://github.com/prusa3d/PrusaSlicer/commit/2d428198b14b7025775212264ddba68ca20b5d7a

    The issue was handling of the special "default_filament_profile" vector,
    which is not being normalized in length to the number of extruders,
    as this vector is being shown to the user at the Printer dependencies page,
    and we don't want to present empty fields there, if the default filament
    profile was not defined in the system preset.

lukasmatena, 2021-01-04:
@sej7278 @sp847242 and anyone else suffering from the issue: Could you guys please test the following build and report if it fixes the issue for you? Thanks.
[PrusaSlicer-2.3.0-rc2+7-asan-g2d428198b-202101041021](https://drive.google.com/file/d/1RO4ogcoVaLnkQBkvoi0ekvdIFFyLC3GV/view?usp=sharing)
[PrusaSlicer-2.3.0-rc2+7-linux-x64-asan-g2d42819-202101041021](https://drive.google.com/file/d/1-TKi7NTo6X8iXQMiyWiF_slfg5tneJfD/view?usp=sharing)
[PrusaSlicer-2.3.0-rc2+7-win64-g2d428198b-202101041021](https://drive.google.com/file/d/1ChZrPg3xhtbH3CPz89glXVkibUnUWRE4/view?usp=sharing)

sej7278, 2021-01-04:
@lukasmatena just built from git HEAD and it fixes the issue for me (even without my `; default_filament_profile = ""` workaround) no more crashing and i noticed it shows the print/filament/printer profiles the 3mf file was based upon which is very useful thanks!

[image]

sej7278, 2021-01-04:
your new build fixes the problem on macos bigsur too.

well, i never tried an earlier version, but it doesn't have a problem with any custom gcode or small 3mf files i throw at it.

sp847242, 2021-01-04:
> 
> 
> @sp847242 Could you please share the content of your PrusaSlicer configuration directory?
@bubnikv The ZIPfile is attached.
Note: This is _after_ I ran the updated version of PS though; I saw that post first.

[PrusaSlicer.zip](https://github.com/prusa3d/PrusaSlicer/files/5767158/PrusaSlicer.zip)

> @sej7278 @sp847242 and anyone else suffering from the issue: Could you guys please test the following build and report if it fixes the issue for you? Thanks. [...]

@lukasmatena I ran this and it looks like it's behaving itself. (win64 version on Win10 Pro)
I tested:
- Open .3MF directly in PS using Open Project.
- Using File --> Recent Objects.
- Drag&drop from Explorer.
- Right-click --> Open With in Explorer and using PS.

Not a single lockup or crash, so this looks good. Thank you. 👍

bubnikv, 2021-01-06:
Fixed with https://github.com/prusa3d/PrusaSlicer/commit/2d428198b14b7025775212264ddba68ca20b5d7a
closing

closed 2021-01-06 by bubnikv