prusa3d/PrusaSlicer #5551: Wipe tower not displayed properly

Kachidoki2807, 2020-12-21:
Hello,

I noticed two bugs regarding the tower when it is not yet computed (with the teeth):
1. The wipe tower isn't displayed at startup while the current print profile has it enabled. To show it up, we need to select another profile and go back. We can also add a part to make it appears, but when removing it, the wipe tower stay displayed. There is a small inconsistency on the conditions to let appears the wipe tower.

> Step to reproduce:
> - Open PrusaSlicer with a stock MMU2S profile
> - There is no wipe tower
> - Add a part or switch to another profile
> - There is a wipe tower
> - Remove the part
> - There is still the wipe tower

2. The wipe tower depth is 'randomly' calculated over the width. I mean, if you play with the width value, decrease it let say to 10mm, and then increase it back to 60mm, the tower is completely out of size. Moreover, repeating this action with different values doesn't give the same results when going back to a known value.

> Step to reproduce:
> - Open PrusaSlicer with a stock MMU2S profile
> - Go to Print settings -> Multiple extruders -> Width -> Change it to 10mm
> - Check that it is OK
> - Go back to the settings and revert the width value
> - Now the tower is huge

You can try multiple time with different values, the results are different.

I noticed that last one because on my IDEX profile with only two extruders, a 10mm width tower and adapted purging volume, I got a very depth tower.

lukasmatena, 2020-12-23:
Thanks, reproduced. The first issue is new in 2.3, the second one is already present in 2.2 (I'm quite surprised it was not reported until now). The issues seem harmless (it is only the toothed-wipe tower that is not displayed correctly, slicing is fine). The fix can probably wait for 2.4 release.

Kachidoki2807, 2020-12-24:
You are right, it is only cosmetic. However I feel that there is something else that is wrong underside.

I mean, I have the feeling that the depth (which I assume is the max depth with 5 extruders) of the toothed wipe tower is arbitrary and based on the MMU, with a relativly big purge volume. Changing the Purging volumes have no effect on the virtual max depth, but changing the number of extruders does.

The issue is that when I use a two extruders setup, with a 10mm wide tower and 10mm³ of purge instead of 140mm³ and a minimum of 15mm³ in the filament settings, the wipe tower which should look like to a square of ~10mm is more looking like to a rectangle of 10x100mm. The side effect is that when slicing, I have the red background because there is an object outside the print area, which is not true.

Size at the opening of the printer profile:
[image]

Size when importing an STL (I noticed that any action on the plater make the wipe tower instantly huge, and the undo command have no effect on its size):
[image]

Red backgroung during slicing, an then the slice result:
[image]

sidekickc, 2022-06-23:
This is still happening with the latest version of PrusaSlicer. Is anyone updating this software?!

stepikovo, 2022-06-27:
@sidekickc as you can [see](https://github.com/prusa3d/PrusaSlicer/releases), the SW is getting updated constantly, thank you for your patience

closed 2026-09-29 by prusaslicer-bot[bot]