prusa3d/PrusaSlicer #5552: Bug: Models transpose off supports after hollow and drill with 3mf files

riutortmd, 2020-12-21:
### Version
Version: 2.2.0+
Build: PrusaSlicer-2.2.0+-202003211132

### Operating system type + version
Mac OS 10.15.7
Windows 10 64bit
(bug occurs in both operating systems)

### 3D printer brand / version + firmware version (if known)
Prusa SL1

### Behavior

1) Import 3mf file.  
2) Slice model with supports, model will be correctly displayed in preview and 3d editor views.
3) Add hollow or hole to model using hollow and drill (happens with either or both operations on various 3mf models)
4) Re-slice model, model will be correctly displayed
5) Switch to 3D editor view from preview view... model with come off the supports and be incorrectly transposed
6) This behavior does not occur with STL files, only 3mf 

Model on left is 3mf, on right is STL.  Before hollow or hole operation
[image]

After hollow or hole operation
[image]

Screen recordings of bug creation

https://user-images.githubusercontent.com/73485493/102801074-b8ca3080-4382-11eb-87b3-5b1b2329a470.mp4

https://user-images.githubusercontent.com/73485493/102801090-bcf64e00-4382-11eb-8f3e-909e0a26dbce.mp4
#### Project File (.3MF) where problem occurs

[Prusaslicer_3mf_hollow_bug.zip](https://github.com/prusa3d/PrusaSlicer/files/5725462/Prusaslicer_3mf_hollow_bug.zip)

_Is this a new feature request?_

No

lukasmatena, 2020-12-21:
We fixed some issues of this sort in the last release cycle. I did not manage to reproduce it now (but maybe I did something a bit differently). Could you please retest this in [2.3.0-rc1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.3.0-rc1)? Thanks.

riutortmd, 2020-12-21:
Thank you, unfortunately the problem still persists on the 2.3.0-rc1
build.  The model looks fine in the slice preview but transposes itself
once the view is switched back to the 3D editor. The model on right is the
STL file on the left is the 3mf.  I have attached the slicer file as well
as the model files separately.  Regards.

P.S.  i will attempt to reproduce on a different system.  I am running dual
boot OSX/Win10 on the same hardware.

[image: Screen Shot 2020-12-21 at 1.55.58 PM.png]

[image: Screen Shot 2020-12-21 at 1.52.55 PM.png]

lukasmatena, 2020-12-21:
Thanks. I paid more attention this time and I can see the problem. We will fix this, but not for 2.3 release.

closed 2026-09-29 by prusaslicer-bot[bot]