prusa3d/PrusaSlicer #5553: adding registration marks during cut

rmeden, 2020-12-21:
### Version
2.2.0

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Prusa i3 mk3

### Behavior
When cutting an object to provide a flat joining  surface (avoiding supports)  you have to manually match the two surfaces perfectly.

New Feature

Optionally add "registration marks" to allow two halfs to lock together for gluing.

Of course you don't want a positive object sticking up marring your flat surface, but how about a (1.75/2)mm rectangular depression on both faces?  (A small bridge).  You could then cut a short section of filament and insert it before gluing.  Probably have at least a X and Y orientation.

#### Project File (.3MF) where problem occurs
N/A

neophyl, 2020-12-21:
Haven’t checked with the most recent versions but you can do this already. If you add modifiers in the middle of the part that have no infill etc and then perform the cut you end up with 2 sets of modifiers that match up in both parts. If you make the modifiers cylinders the size of pins you want to use you can use them to key the parts back together for gluing.

rmeden, 2020-12-22:
Just tired your suggestion, but it didn't work. 

I could see the cylindrical hole when slicing the main object, but if I tried the cut in the middle of the cylinder, the cylinder always stuck with one of the cuts.

It was also quite tricky and not obvious to get close.  When you rotate the cylinder to be horizontal, the Z position isn't used to position the cylinder. :(  It also seems to be a delta from the top of the object...

I'm looking for a simple option during a cut you don't need to be a prusaslicer expert to get done.  It could be implemented using modifiers (if it worked), but the operation  should be pretty point-and-click.

lukasmatena, 2020-12-22:
We are considering to add this feature for some time already, it will likely get implemented sometimes.

Kachidoki2807, 2020-12-22:
@neophyl's idea is a good starting point, I tried my-self on 2.3.0-rc1, and it need a small workaround to avoid the issue of @rmeden.

When the modifier is placed properly, you need two more steps before to cut the object. First, add an instance of the object (including the modifier), second set the copy as a separated object. Now it is possible to use the cut tool by setting keep upper part on one instance, and keep the lower part on the second instance.

I admit that it is a bit tricky, but doable. While waiting for better.

foreachthing, 2020-12-24:
@rmeden, maybe this works (easier):
1. Import model
2. add modifier(s)
3. copy & paste part (CTRL+C/V)
4. perform cuts on both models at the same height (or with compensation for glue, etc.). For one model keep top the other keep bottom half.

rmeden, 2020-12-27:
@foreachthing that was tried above.  There were issues on both the current and next version. @Kachidoki2807 found a way to make it work, but the whole process is much too complicated.  I'm proposing a point and click solution on the cut operation, that could get implemented with modifiers.

neophyl, 2020-12-27:
I just tried the original method I mentioned in rc2 and it still works.  One of the ,imitations on the modifier(s) you place inside the object where you want the registration void is that it must use a different perimeter count to the main object and zero infill of course, you also need to set bottom layers to zero too, otherwise you dont get internal walls just the infill void.

Basicallly you slice with the modifier first to check its creating a nice-ish internal hollow and then when you perform a cut with keep top and bottom and rotate bottom to face bed the modifier also gets duplicated and you end up with the void in both parts.  

I agree that it is difficult to place the modifier(s) which is why I requested an x-ray mode ages ago but I don't use the technique any more anyway as if you want changes to the model then you really should be modifying the model in the first place with software that is designed to do so.  Yes it can work but its like trying to hammer in a nail with a knife.  Far more effort that just using the proper tool for the job required.  

Slicer is great but I really cant understand why people want it to be able to do everything.  Its a slicer, let it be the best slicer it can be and keep the modelling tools for modelling software.

rmeden, 2020-12-28:
> Slicer is great but I really cant understand why people want it to be able to do everything. Its a slicer, let it be the best slicer it can be and keep the modelling tools for modelling software.

When I do a design, I do try to make things easily printable.   In the case that triggered this feature request, I was trying to print a sword from Thingiverse where the designer didn't make it easy to print (lots of shallow detail on both sides, supports didn't release)  and didn't have a license anymore for the program he used to make the design.

I think the "cut" feature in prusaslicer is designed exactly for cases like this (cut and flip), so adding the new feature would make that feature more usable.

bubnikv, 2021-06-03:
PrusaSlicer 2.4.0-alpha1 will implement "negative volumes" aka "subtractors" as a new type of object volume, thus not requiring the "no perimeters, no bottom / top layers" workaround.

The pegs / holes will be implemented in some of a further version though.

Diminik6280, 2022-12-19:
Bonjour à tous,
Oui, je sais, le sujet a déjà été traité plusieurs fois, mais je ne sais pas ou répondre....
Je ne suis pas un grand technicien, mais je vais vous faire part de mon expérience et de mon avis quand même...;-)
Le fait de pouvoir ajouter un volume négatif afin de créer une mortaise cylindrique de 1.90mm est déjà très bien, mais uniquement si on doit découper la pièce... Pour les pièces déjà découpées, ça ne fonctionne pas... comme par exemple ce modèle [https://www.thingiverse.com/thing:5586541/files](url). ce qui est bien dommage car pour le découper soi-même, il faut le réassembler dans un logiciel à part, le ré-importer dans Prusaslicer et le découper à nouveau avec tous le boulot que ça représente.... De plus, Je n'ai pas trouver la possibilité de rendre la pièce translucide afin de placer le petit volume négatif de manière précise... Donc, oui, la demande pour l'implémentation d'une fonction dédiée dans Prusaslicer est fondée, et je dirais même indispensable..... Merci à la communauté d'insister auprès de PRUSA...;-)

lukasmatena, 2023-02-01:
Implemented in [2.6.0-alpha2](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha2). Closing.

closed 2023-02-01 by lukasmatena

Diminik6280, 2023-03-23:
Great !!! it really changes all the interest of the cut!... But to go a little further, it would be extraordinary to be able to make cuts with a "clips" function which would dispense with the application of glue and which would also be clippable and can be unclipped... A bit like on the attached sketch...
[image]