prusa3d/PrusaSlicer #5554: [Feature Request] 'Active Temperature Control' for multi-extruder prints: dynamically heat/cool tools so they are ready when needed.

n8bot, 2020-12-21:
PrusaSlicer has very accurate print time calculation built in. Using this ability, it should be fairly simple to implement a feature to cool down and heat up tools so that they are ready for use when they are needed. The feature is intended to reduce the time taken waiting for tools to heat up at tool changes.

The slicer IceSL has this feature. They describe it like this:

> Active temperature control is specifically designed to avoid material to degrade and ooze from idle extruders. The idle extruder is dynamically cooled down and heated during printing, such that it is ready when needed and cooled when not needed. Adjust extruder parameters idle_extruder_temp_degree_c and extruder_degrees_per_sec when using this feature.

IceSL implements this feature well. A target standby temperature is set very low: 100 degrees C by default. The user then estimates and inputs the time it takes to heat/cool in degrees per second. The slicer then calculates, for all tool changes, at what point during the layer to start pre-heating the next tool. Also, importantly, it decides to what temperature to cool the tool being deselected so that it doesn't cool down too much for the time it needs to be inactive. This means the tools may be commanded to different temperatures for standby depending on how long it will be inactive. It gets as cool as possible, given the time, and with enough time will be cooled down much lower than other slicers achieve.

This is only really useful for printers with discrete tools: IDEX or toolchangers. Prusa may not have any such printers on the market right now, but they should make an IDEX printer because they are friggin rad.

Also, if Prusa Research is not willing to implement this feature, I would be glad to help in any way I can, and seek help form others to implement this feature in a way that would be acceptable to the Prusa team.

Kachidoki2807, 2020-12-22:
Duplicate https://github.com/prusa3d/PrusaSlicer/issues/5248

n8bot, 2020-12-22:
> Duplicate https://github.com/prusa3d/PrusaSlicer/issues/5248

Oops! Sorry about that. I tried searching and couldn't find that one. I should have looked under the multi-material tag rather than searching. 

I'm glad to know I'm not alone in wanting this!

Have you tried IceSL? How do you like they way they do this? 

When they close this for being a duplicate I will re-add my comments of support to your issue.

Kachidoki2807, 2020-12-22:
> Oops! Sorry about that. I tried searching and couldn't find that one. I should have looked under the multi-material tag rather than searching.

No problem. ;-)

> Have you tried IceSL? How do you like they way they do this?

I have tried IceSL a long time ago, but unfortunately not under this aspect.

But I tried this feature from IdeaMaker, and what I remember is that the temperature reaching is not awaited at the toolchange. So when going to some extreme deltas, I got a cold extrusion error. But that's not the worst, I also got a printer kill heating failed, because under some conditions, the PID never stabilize cause the temperature setpoint constantly change.

I have also gave a try to the ooze prevention from PrusaSlicer, not compatible with wipe tower and imply to use an ooze shield. This one wait for the temperature, but the ooze shield is the worst I have ever tried. The nozzle wait few millimeter outside the ooze shield, which make it ooze a lot (!) and then the shield is supposed to grab the ooze... sometimes.

So it is not trivial to implement, but with a wipe tower to help, it might be doable.

bubnikv, 2021-01-05:
We plan to do exactly that: To use the time estimator to plan for the extruder heat up. It may be part of PrusaSlicer 2.4.

n8bot, 2021-01-05:
Fantastic. Thank you.

Starscream83, 2021-04-02:
Hi, any news about this topic? Will this feature be present in 2.4?

n8bot, 2021-11-22:
@bubnikv how are the chances looking that we will see tool pre-heating for toolchanges in 2.4.0?

I know I am not the best coder, but I would be willing to volunteer time if needed to test this feature or anything, to expedite it's arrival in prusaslicer.

Also, would be cool if custom toolchange/start/end g-code is included in time estimation if it's not already. Even just the ability to add a user-determined number of seconds for specific events would be good.

bubnikv, 2021-11-30:
We plan to implement extruder preheat to support our upcoming XL printers.

n8bot, 2021-11-30:
Lol, yes, I guessed that. Hence my offer for help testing it! Just an offer.

n8bot, 2021-12-04:
@bubnikv would it be possible to allow custom g-codes to manually adjust the estimated amount of time required to execute?

For instance, in my tool change G-code I call a custom .gcode file on my printer's sdcard to park the tool. 

```
M98 P"tpark{previous_extruder}.g" ; park tool
```

The time estimator probably thinks this takes zero time, but maybe it takes an average amount of time, or an exact time, that the user can determine. Then maybe we could take the line with the amount of time to add (or subtract? who knows) from the time estimate. I.e.:

```
M98 P"tpark{previous_extruder}.g" ; park tool [time_adjust=20]
; or
M98 P"tpark{previous_extruder}.g" ; park tool [time_adjust=-20]
```

traxman25, 2025-03-04:
This would make Prusa Slicer a clear winner again in multi tool printing. It is the single reason I stay with other slices even though they are FAR more buggy than Prusa Slicer. A simple Preheat Idle Extruder Time such as Orca Slicer uses works just fine.

closed 2026-09-29 by prusaslicer-bot[bot]