prusa3d/PrusaSlicer #5557: Brim Issue

gryphonB-2, 2020-12-22:
### Version
2.4

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
Kubuntu 18

### 3D printer brand / version + firmware version (if known)
Not part of issue
### Behavior
The issue lies with the way slic3r creates brims. In FreeCAD I created a spoked wheel, oriented vertically, then meshed it. I dragged it into slic3r where it was displayed vertically. I rotated it 90 deg to print flat on the plate. 

On adding Brim the brim was only applied in a region about the original bottom of the part - not around the entire new perimeter. The brim also was applied inside the perimeter of the part. The picture tells all.

To test, in FreeCAD I rotated the part 90 deg to lay it flat, then meshed again & again dragged into slic3r. This time the brim extended around the entire perimeter, as expected & desired.  
One rather odd thing: With the meshed-while-horizontal part already sliced, I then imported the meshed-while-vertical part and rotated it 90 deg to lay flat. On slicing, the brim was applied to entire perimeter on both.    

And on another note, what's with all the 'we need to verify who you are, frankly we'd appreciate it if you were microchipped' hoopla? Is this Microsoft trying to Facebook everyone? Normally, I wouldn't put up with this BS. But this is an issue you need to be aware of. GF MS. 

[image]

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

neophyl, 2020-12-22:
Check your first layer is actually all on the plater. Usually that is a sign that its not actually flat to the bed.   Or instead of rotating it 90 degrees use the place on face tool.

Kachidoki2807, 2020-12-22:
> ### Version
> 2.4

Welcome back from the future.

> On adding Brim the brim was only applied in a region about the original bottom of the part - not around the entire new perimeter. The brim also was applied inside the perimeter of the part. The picture tells all.

No, it isn't telling us anything. Please enclose a .3MF of this very specific case, or at least the STL, thanks. What happens if you explore the first layers, as mentioned by @neophyl?

Are you able to check this behavior with another slicer, as a first level analysis, it could involve your STL instead of the slicer. Your FreeCAD model can be perfect, but the STL export can always introduce some alignment defects due to the meshing process.

> And on another note, what's with all the 'we need to verify who you are, frankly we'd appreciate it if you were microchipped' hoopla? Is this Microsoft trying to Facebook everyone? Normally, I wouldn't put up with this BS. But this is an issue you need to be aware of.

Huh? I really don't know what you mean.

closed 2026-09-29 by prusaslicer-bot[bot]