prusa3d/PrusaSlicer #5558: Printer randomly forcing filament change

123killer123, 2020-12-22:
### Version
2.3.0. -beta3

### Operating system type
Windows 10

### 3D printer brand / version + firmware version (if known)
Original Prusa i3 MK3S (3.9.2-3524)

### Behavior
Printer is randomly forcing filament change mid print. Firstly it happened mid print and I tought that it was my error but then it happened again  3 times in a row on the same layer height. After reinserting the filament 3 times it started to print normally but after 10 minutes same story. I am using octoprint and have managed to copy few lines of communication between printer and octoprint (it looks like that the printer enforced M600 aka. filament change). This happened on two different models but NEVER happened in any older version of Prusa slicer (I have managed to print same print sliced in v. 2.2.0). In .zip file I added gcode of model and few copied lines between printer and octoprint. Let me just restate that everything worked everytime with older slicers so I doubt octoprint is to blame this time :).

#### Zip
[Slicer error.zip](https://github.com/prusa3d/PrusaSlicer/files/5731641/Slicer.error.zip)

rtyr, 2020-12-22:
It is filament change triggered by filament sensor (possibly false filament runout detection). You can disable filament sensor through LCD menu to test it.

Closing, not PrusaSlicer related issue.

closed 2020-12-22 by rtyr

123killer123, 2020-12-22:
> It is filament change triggered by filament sensor (possibly false filament runout detection). You can disable filament sensor through LCD menu to test it.
> 
> Closing, not PrusaSlicer related issue.

Dear. I was thinking that this is not related to slicer at all but when resliced with version 2.2.0 print completed succesfully without problems...

rtyr, 2020-12-22:
It may be caused by some specific vibrations or something like that, but you can see from your terminal output, that M600 was triggered by filament sensor.

[image]

123killer123, 2020-12-23:
Ok. You are the expert. Thank you for the advice.