prusa3d/PrusaSlicer #5560: Object sliced with errors

hytakara, 2020-12-23:
### Version
PrusaSlicer-RC2+win64
### Operating system type + version
Windows 10-64 (build 19042)
### 3D printer brand / version + firmware version (if known)
Anycubic Chiron v1.3.0
### Behavior
object sliced with errors
import STL file
configure parameters
generate G-code
sliced object similar to the original.
sliced object is different from the original.

[image]
[image]
[image]
[image]

[config.zip](https://github.com/prusa3d/PrusaSlicer/files/5733015/config.zip)

rtyr, 2020-12-23:
The walls of the object are too thin to be printed with your extrusion width.

Please provide your 3MF project or at least STL (you attached only config.ini).

hytakara, 2020-12-23:
Sorry, you are correct, I compared the advanced settings and they were different resulting in the problem.

STL File uploaded
[fan_test_-_Part_1.zip](https://github.com/prusa3d/PrusaSlicer/files/5735475/fan_test_-_Part_1.zip)

closed 2020-12-23 by hytakara