prusa3d/PrusaSlicer #5561: 2.3 rc2 still localisations missing (German)

hilleke-dirk, 2020-12-23:
### Version
Version 2.3.0-rc2+win64

### Operating system type + version
Win 10

### Behavior
still some points untranslated

Minimum shell thickness ...
Avoid crossing perimeters ...
Lenth ot he infill anchor ...
Ironing ...
Extrude Clearance ...
several keyboard shortcuts

YuSanka, 2020-12-23:
Thanks for your report, but can't reproduce your bug 
[image]

hilleke-dirk, 2020-12-23:
I just copied the content of the new zip in the installation folder.
Than I had this effect.
After deleting everthing and a new copy everything is fine now.
Thank you.

closed 2020-12-23 by hilleke-dirk