prusa3d/PrusaSlicer #5566: Preview rendered invisible

umtauscher, 2020-12-23:
### Version
Version 2.3.0-rc2+

MacOS Sierra
10.12.6

### 3D printer brand / version + firmware version (if known)
Anet Et4 Pro 1.0.2

### Behavior
When slicing a model the preview vanishes, as soon the gereating G-Code is finished
The gerated G-Code seems to be generated correctly though.
I used thios model:
https://www.thingiverse.com/thing:4239135
Clip Standard and Clip Large both show the same problem.
Simpler designs seem to work better
[image]
[image]
[3D4U_Coffee_Clip_Standard_V2.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/5735060/3D4U_Coffee_Clip_Standard_V2.3mf.zip)

rtyr, 2020-12-23:
I am unable to reproduce it with 2.3.0-rc2 win64

Area5142, 2020-12-23:
@umtauscher 
It looks like the issue #4719 I had on Linux Mint 19.3 x64 with PS 2.3.0 alpha.
Preview was first missing and when forced redrawn it looked like second screen dump.

umtauscher, 2020-12-24:
It looks tu me like some internal variables are not properly initialized....

K5HJ, 2021-01-03:
I also have the same problem with win64 version. Large objects will not show in preview after slicing. Smaller objects will.

umtauscher, 2021-01-03:
Glad you can reproduce it.

bubnikv, 2021-01-04:
@enricoturri1966 claims that he fixed this issue before with https://github.com/prusa3d/PrusaSlicer/commit/36bedfdebfa1c82535c469b251c397b88c260ce9

@enricoturri1966 would you please look into this issue?

bubnikv, 2021-01-04:
> MacOS Sierra 10.12.6

That is quite an old model, right? What GPU? How much RAM do you have?

bubnikv, 2021-01-04:
@umtauscher Does PrusaSlicer 2.2.0 work correctly on the same machine?

umtauscher, 2021-01-04:
@bubnikv 
Hi,
yes ist’s old. Nevertheless I thought it was reproducable on Win64 as well?
My machine has 16 GB of RAM, I5, Intel HD3000 grafics.
I cannot tell, if it worked on 2.2 because I cannot test it any more. The profiles are incompatible.

Silvertips, 2021-02-18:
Hello, i have the same problem : artefacts or object invisible on a MacBook Pro (13 inchs, early 2011) 8 gb ram 2,3 GHz Intel Core i5 whith High Sierra.
version of Prusa slicer : 2.3.0

I wanted to let you know .

Ask me if you need more details

Cheers

enricoturri1966, 2021-02-19:
Could you please test if the following build (taken from current master) helps in fixing the issue ?
https://www.dropbox.com/s/q4i4gf1fn2cerin/PrusaSlicer-2.3.0%2B278-universal-gfbd8d57ad-202102190006.dmg?dl=0

Silvertips, 2021-02-19:
> Could you please test if the following build (taken from current master) helps in fixing the issue ?
> https://www.dropbox.com/s/q4i4gf1fn2cerin/PrusaSlicer-2.3.0%2B278-universal-gfbd8d57ad-202102190006.dmg?dl=0

Tested and so far it seems that the problem is fixed 👍 

Thank you.

umtauscher, 2021-02-19:
Hi Silvertips,
I just tested the new version using the same environment and files and it seems to work just fine.
Thank you.
Cheers

paperra, 2021-03-04:
Hi all

Just wanted to add that I've had the same problem since upgrading to 2.3.0 (snapshots below). It's sporadic and seems to occur most frequently on prints with circular layers. I'm using a Macbook Air with 4GB RAM, 1.7GHz i5 processor running High Sierra and and PrusaSlicer-2.3.0+-202101111304. No problems with previous versions of PrusaSlicer.

@enricoturri1966 I'll test out the build you suggested to Silvertips.

[image]
[image]

Thanks

bubnikv, 2021-10-05:
The OSX issue was fixed in PrusaSlicer 2.3.1.

closed 2021-10-05 by bubnikv