prusa3d/PrusaSlicer #5568: Bug: Unselecting more than 10 copies stops at copy 10

bartrail, 2020-12-23:
### Version
Version 2.3.0-rc1+win64

### Operating system type + version
Windows10 

### 3D printer brand / version + firmware version (if known)
mk3s stock / 3.9.1

### Behavior
* Unselecting copies from the printbed only works in 10/20 copies 
** Download the *.3mf files from this project: https://www.prusaprinters.org/prints/49435-ear-savers/files
** Import any of the 2 *.3mf files to PrusaSlicer and unselect the copies on the right side
** After unselecting copy 11 the copy 10 will be selected again. All other copies 11-20 just toggle copy 10.
* Expected: Unselect all copies like in Version 2.2.0 (also works in Version 2.3.0-beta2)
* _Actual Results_
** [image]

#### Project File (.3MF) where problem occurs
https://www.prusaprinters.org/prints/49435-ear-savers/files

bartrail, 2020-12-25:
Update: with RC2 the problem remains, but it starts at copy 16 instead of 10.

bubnikv, 2021-01-08:
This was a long standing issue, which we were able to reproduce in PrusaSlicer 2.2.0 as well.
It will be fixed in PrusaSlicer 2.3.0-rc3.

closed 2021-01-08 by bubnikv