prusa3d/PrusaSlicer #5570: Feature Request: Paintable Ironing

bonjipoo, 2020-12-23:
I feel paintable ironing would be a nice way to select which top surfaces to iron rather than only being able to select between all top or only top surfaces.  I suppose it would help save time if areas which didn't need ironing could be avoided on the same layer as one that do.

Kachidoki2807, 2020-12-23:
Hmmm, since ironing only apply to flat surfaces (on Z axis), maybe it could be smarter to "click-select" the faces/layers to be ironed?
I mean, a simple mouse click to select a whole face is a lot quicker than manually painting it. (I guess that it is not so easy to handle cause the STL mesh).

Whatever, I like any idea that make the ironing easier to use.

bonjipoo, 2020-12-23:
Yes I like the idea of individually selectable faces also but painting is already a feature of the slicer and might be easier to implement. Its not for me to decide.

rtyr, 2020-12-23:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/5116. Closing.

closed 2020-12-23 by rtyr