prusa3d/PrusaSlicer #5571: No longer possible to print to Octoprint directly

TD-er, 2020-12-24:
### Version
- 2.3.0-rc2

### Operating system type + version
- Windows 10

### 3D printer brand / version + firmware version (if known)
- Prusa mini
- Prusa Mk3S

### Behavior
Just ran the RC2 of the new Prusa Slicer, which imported my present settings of Prusa Slicer.
In there I had both printers with Octoprint API key configured.
However I now am no longer able to send directly to the printer and to make matters worse I can't find where to setup the host + API key to print to.

Also started the previous version of Prusa Slicer and that one was no longer able to read my settings, so I hopefully have them somewhere.

neophyl, 2020-12-24:
You need to add a physical printer now.  Those settings have been split off from the usual printer profile settings.  Its all in the release notes for the earlier 2.3 releases (from the first alpha releases) 
https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.3.0-alpha1
cant seem to get a linkl to take you to the exact spot on the page but its towards the bottom called 'Physical Printers settings'

TD-er, 2020-12-24:
Hmm I added new "physical" printers and noticed the previously set Octoprint settings were present.
So that's positive.
However when I switched in the default view to another printer, the application crashed and was closed.

[image]
Switching from the MK3S to the Mini -> crash

Could not reproduce it though, but to me it looks like the settings parser may benefit from an extra try-catch block.
Maybe Prusa Slicer also tried to connect to Octoprint and the Octoprint instance did not reply in due time?

So this does seem something that would be nice to be fixed before the release.

lukasmatena, 2020-12-24:
From the 2.3.0.-rc2 change log:
> When first installing PrusaSlicer 2.3.0-alpha1, the network connection information is removed from existing user Printer profiles and converted to new Physical Printers profiles giving them generic names "Printer 1, Printer 2, ...". The user is advised to review and rename these Physical Printer profiles generated by the conversion process. 

The application should have created the physical printer profiles for you and tell you about it when you first ran rc1 or rc2 for the first time. Has it not happened? Your confusion would indicate so, but then I don't understand the 

> I added new "physical" printers and noticed the previously set Octoprint settings were present.

Where were they present if you had to add the physical printer profile yourself? What exactly has happened? This information might help us reproduce the crash. Thanks.

TD-er, 2020-12-24:
Just to make sure those 2 will not lead to false conclusions.
Initially I ran the RC2 build of Prusa Slicer on my desktop and today I tried it again on my laptop, just to see what would happen what I may have missed.
I still have to see on my Desktop PC to see whether it may still have the old settings converted without me realizing it.

The crash I reported was on my laptop, which also sometimes has a sub-optimal WiFi connection when I am downstairs in the livingroom. (that's why I also mentioned network timeout as a possible cause for the crash when switching printer profiles)

I will try to get a bit more clear report when I am at my Desktop PC again.

TD-er, 2020-12-24:
OK, I got back at my desktop PC and had a good look at the printer options.
I had not selected the "physical printer", but when I did I still was not able to directly send to Octopi.
So I went into the settings, noticed the Octopi parameters were there, saved them without changing anything.
Then after slicing a printjob, I got the icon again to send it to Octoprint again.

So the good news is, it is working, the bad news is, I have no idea how to reproduce it.

The only things I can think of:
- Maybe I've had the old prusa slicer open when first starting the new RC (I closed it while I started the new one)
- One of the Octoprint instances was printing all day, so maybe this triggered some unexpected conditions when PrusaSlicer queried it?
- The crash happened when switching from the (busy) printer connected to the active Octoprint to the inactive mini on the second Octoprint (each running on a separate Pi)

This last one sometimes is a bit harder to reach also via the web interface when it wasn't accessed for a while. Not sure why as I don't address them via their .local address, but either directly via IP or a DNS entry which is resolved by the router.

lukasmatena, 2020-12-25:
So
- The old config was correctly transformed to the new one and added physical printer profiles, which were automatically named such as "Printer 1 * Original Prusa MK3S - Copy" and similar
- You then manually renamed the physical printer profiles to "Prusa MK3S * Original Prusa MK3S - Copy" which can be seen on your screenshot
- Everything is fine now
- You once missed the Send to Printer button when the Octoprint was busy and got a crash in the same scenario after switching profiles

Is that about right? It is possible that there is some problem when the OctoPrint is running (maybe even present in 2.2.0?). At this point I'd like to be sure that the conversion of the profiles (which is new in the 2.3 series) is not the problem.

TD-er, 2020-12-25:
The conversion of the profiles does seem to be OK (just tested on another laptop which I had not used for a while)

The moment of crashing was not when I tried to send to the printer, but just by switching the printer profile.
Right after clicking in the combo box to select the other profile, the application stopped to respond and crashed.

YuSanka, 2020-12-29:
Can you please send us your config bundle?
[image]

TD-er, 2020-12-29:
As my instances of OctoPi are not reachable from anywhere (not even from my guest network) I left them in the files to keep them as untouched as possible.
[Gijs_PrusaSlicer_config_bundle.zip](https://github.com/prusa3d/PrusaSlicer/files/5752117/Gijs_PrusaSlicer_config_bundle.zip)

The config from the laptop is from the machine that had the crash of Prusa Slicer I mentioned.

YuSanka, 2020-12-30:
I still don't observe the crash on both of your config bundles. :(

TD-er, 2020-12-30:
Last 2 days I've been a bit more critical when using both Octopi instances.
- Octopi.lan -> MK3S
- Octomini.lan -> Prusa Mini

The first one is connected via ethernet and the second one via WiFi (ethernet doesn't seem to be working on that one, no idea why)
The one via WiFi does sometimes react quite slow if it was not accessed for a while. For example the web interface does only load after several seconds.
Probably related to some power saving mode of the WiFi (not looked into yet).
If the device was not actively accessed for a while, it is very well possible the ARP tables in the switches etc. were no longer aware of the device and then it may take a while for the device to become reachable again, when it does reply again on ARP requests.

Today I noticed that node did give a timeout popup in Prusa Slicer and also the web interface did take way longer then normal, but after some 10+ seconds it showed and Prusa Slicer could also reach it.
This means Prusa Slicer does use timeouts etc. 
But what if the device was first found and later when switching between printer definitions it doesn't reply?
Is Prusa Slicer using a different timeout then? 
Or maybe the reply from the network stack differs when the host knows the MAC of the remote, but the network can't handle it?

Just like when you send pings to a host, which has a flaky connection or is rebooting. You then also get different replies based on where in the network(stack) the failure of the ICMP packet occurs.

bubnikv, 2021-01-04:
I personally doubt that the crash has anything to do with the actual connection. When switching profiles, no OctoPrint communication is triggered. When pressing the Test button at the Physical Printers dialog, the test is blocking. After the test finishes, no communication is running (at least this is the intention).

A similar issue https://github.com/prusa3d/PrusaSlicer/issues/5658 may or may not be related to yours.

To me it looks like a result of conversion of the profiles, thus once the profiles are converted, nobody will be able to reproduce the issue with the converted profiles. It would really help, if you could reproduce your issue if you have a copy of the original profiles. What surprises me is that you are the only one reporting this issue, so the issue may really be related to your profiles before conversion.

It may have something to do with the PrusaSlicer 2.2.0 running at the same time, but they should only write the PrusaSlicer.ini which may lead to some incompatibilities, but PrusaSlicer 2.2.0 should not save any of the other profiles (print, filament, printer) without user clicking the "save" button.

How many times did you experience the crash? Did you ever experience the crash after the profiles were converted and after you restarted PrusaSlicer?

TD-er, 2021-01-04:
It happened only once, on the laptop. (thus WiFi connection, although it may not be related)
It did not happen on the PC, nor on the other laptop which was connected to LAN.

I will have a look at another laptop which is not in use anymore, to see if that one has a profile stored in the old format.
And maybe in backups an old profile may be present.

bubnikv, 2022-06-29:
This is a stalled issue reporting a difficult to reproduce error on old build.
Closing.

closed 2022-06-29 by bubnikv